w3c/automotive #2: JavaScript coding style is not enforced

djensen47, 2015-05-05:
Frequently throughout the document JavaScript coding guidelines are not used. For instance, curly braces are found both at the end of the line and also on it's own line. 

There is no current JS style guidelines that uses curly braces on a new line.

Options for coding guidelines:
- https://google-styleguide.googlecode.com/svn/trunk/javascriptguide.xml
-  http://javascript.crockford.com/code.html
- JavaScript Standard Style https://github.com/feross/standard
  - This one is new and maybe controversial because it says not to use semicolons.

The only change I would make it for 2-space indentation instead of 4. Those extra spaces might push code samples out of view, which is annoying to say the least.

astorije, 2015-05-06:
For what it's worth, [w3c/echidna](https://github.com/w3c/echidna) enforces coding style using [JSCS](http://jscs.info/) and an [extensive set of rules](https://github.com/w3c/echidna/blob/master/.jscs.json).
I am not sure how to enforce this on a spec, but it might be worth giving a try.

djensen47, 2015-05-22:
- [ ] See what other groups use
  - Looks like JSCS
- [ ] Decide on coding style
- [ ] Editors (or others) to provide pull requests to enforce style guide
- [ ] Optional - find a tool to make finding errors eaiser

djensen47, 2015-05-26:
Since w3c/echidna is for the new publication workflow, it seems like a good idea to use their JavaScript style guidelines. The rules are provided in the link above provided by @astorije.

closed 2015-07-20 by acrofts84