w3c/automotive #4: Update README

djensen47, 2015-05-06:
The README needs to be fleshed out:
- [ ] Brief introduction about the group and the spec
- [ ] Repo layout: where do files live. This was discussed in a recent meeting but should be documented in the README
- [ ] Style and/or code guidelines. Links are good enough.
- [ ] Contribution guidelines: any instructions on how to contribute and who is allowed to contribute (e.g., members), and CLA requirements, etc.
   - It might be a good idea to have a separate CONTRIBUTORS.md file with a list of folks who have and can contribute. It might make lief easier on the editors.
- [ ] Tools used. E.g., a link to Respec

djensen47, 2015-05-22:
- [ ] @djensen47 to provide a pull request

acrofts84, 2015-07-16:
I've submitted and merged a pull request on this with an update in each section. As we expand I'll make sure to keep it up to date. In terms of contributions, I've linked to the Charter which covers licencing, and those who have contributed can be found <a  href="https://github.com/w3c/automotive/graphs/contributors">here</a> within GitHub.

closed 2015-07-16 by acrofts84