w3c/automotive #5: nullable and non-nullable attribute definition

QingAn, 2015-05-11:
In data spec, http://www.w3.org/2014/automotive/data_spec.html   
for the multiple kinds of attributes defined in different interfaces, some attributes are defined as nullable, like fuelType, which is nullable, in FuelConfiguration Interface (http://www.w3.org/2014/automotive/data_spec.html#idl-def-FuelConfiguration), and VehiclePowerMode, which is non-nullable, in VehiclePowerModeType Interface (http://www.w3.org/2014/automotive/data_spec.html#idl-def-VehiclePowerModeType)

We need a clear explanation why such attributes are set in this way.

djensen47, 2015-05-21:
Action items

- [ ] Check with Kevron (@tripzero) and Justin (@jsbach86) why some are nullable and notnullable.
- [ ] Document the guidelines

tripzero, 2015-05-26:
The general rule I think we've been following is that if the interface is provided, it MUST implement the non-nullable items otherwise it's optional.  Many of the non-nullable attributes are found in interfaces with only one attribute (ie VehicleSpeed).  Logically it doesn't follow to even offer the interface (ie make it available as an attribute on "vehicle") if it has no attributes.

closed 2015-07-07 by QingAn