w3c/automotive #6: Discovery API: to discover the list of vehicle data interfaces

QingAn, 2015-05-12:
Through the Web application running on IVI or user’s mobile device, user can access the vehicle data defined in http://www.w3.org/2014/automotive/data_spec.html. Before that, it is necessary to discover which vehicle data interfaces are accessible to the Web application. This would be essential if the access control policy is implemented, since the accessible vehicle data interfaces may vary from application to application.

One suggested Discovery API design is as follows:

enum APITypeEnum {
    "FuelConfiguration",
    "Identification",
    "SizeConfiguration",
    "SteeringWheelConfiguration",
    "TransmissionConfiguration",
    "WheelConfiguration",
    "Acceleration"
};  //This is just an example. We can add more according to the definition in data_spec

[NoInterfaceObject]
interface AvailableAPIs : VehicleCommonDataType {
    readonly    attribute unsigned short? count;
    readonly    attribute APITypeEnum[]? APIType;
};

APIType of type array of APITypeEnum, readonly , nullable
must return type of API available to access. If the vehicle provides multiple APIs, APIType returns an array of API types.

count of type unsigned short, readonly , nullable
must return total number of APIs available to access

aShinjiroUrata, 2015-05-15:
Hi QingAn-san,
It looks [14. Data Availability](http://www.w3.org/2014/automotive/vehicle_spec.html#data-availability) is for similar purpose( but we need to use 
<pre>availableForRetrieval()
availableForSubscribe()
availableForSetting() </pre>
to check availabiity one by one and not possible to get available list at once).
Do you think this 'Data Availability' is enough or you need something different?

QingAn, 2015-05-15:
The  http://www.w3.org/2014/automotive/vehicle_spec.html#data-availability  just supports the detection of whether an attribute is available or not within a specific interface. 

Discovery is focusing on discovery of available interfaces.

So it's kind of different level between 'Data Availability' and 'Interface Discovery'.

But I think it's a good idea to combine them to realize a finely grained discovery (per attribute in each interface). And if so, a unified data format needs to be defined.

djensen47, 2015-05-21:
Action items:

- [ ] General feedback on @QingAn's suggestions
- [ ] Look for other examples in W3C for discovery
  - Devices API
  - Location?

closed 2015-07-24 by QingAn