w3c/automotive #9: [vehicle_spec] enum type ZonePosition

QingAn, 2015-05-13:
In http://www.w3.org/2014/automotive/vehicle_spec.html#zone-interface  enum type ZonePosition is defined. 

enum ZonePosition {
    "front",
    "middle",
    "right",
    "left",
    "rear",
    "center"
};

But it is never used through the spec.

aShinjiroUrata, 2015-05-15:
In my understanding, standard usage of ZonePosition values are like
<pre>
var zone = new Zone;
zone.value = ["front", "right"];

vehicle.vehicleSpeed.get(zone).then(
  function(vspeed) { console.log( "Current speed is: "+vspeed)},
  function(error) {});
</pre>

So I guess, Zone definition should be written as 
<pre>
interface Zone {
              attribute ZonePosition[] value;  //changed from: attribute DOMString[] value
  readonly    attribute Zone        driver;
  boolean equals (Zone zone);
  boolean contains (Zone zone);
};
</pre>

QingAn, 2015-05-15:
I agree with that. The ZonePosition should replace the DOMString in Zone interface definition.

BTW, an extra comment:
In the example, VehicleSpeed may not be a good example to demonstrate Zone, since vehicle speed would not change in different zones.

Also, for  function(vspeed) { console.log( "Current speed is: "+vspeed)} part, should be changed to:

 function(vspeed) { console.log( "Current speed is: "+vspeed.speed)}

according to VehicleSpeed definition http://www.w3.org/2014/automotive/data_spec.html#vehiclespeed-interface

aShinjiroUrata, 2015-05-15:
Those are my mistakes and you are right.
Thanks for pointing.

djensen47, 2015-05-21:
Action items:

- [ ] Ask Kevron (@tripzero) and Justin (@jsbach86) if this was intentional or an error

tripzero, 2015-05-26:
Agreed that ZonePosition should replace DOMString

closed 2015-06-01 by QingAn