w3c/automotive #12: [data-spec] Datatype consistency - Octet/Byte/Unsigned Short

acrofts84, 2015-05-19:
There seems to be some inconsistency over which data types we use when it comes to smaller numbers. 
>   In the WebIDL specification a "byte" is signed, where an "octet" is unsigned. Therefore we shouldn't use "unsigned byte", as in 10.3.1.
>   "unsigned short" is probably overkill for a 0 - 10 value as in 5.3.1. 
>   As for the difference between octet and byte for small unsigned values it may be arbitrary, but maybe should be consistent.

5.3.1 Attributes
totalDoors of type unsigned short, readonly , nullable
must return total number of doors on the vehicle (all doors opening to the interior, including hatchbacks) (Min: 0, Max: 10)

6.10.1 Attributes
gear of type octet, readonly , nullable
must return transmission gear position. Range 0 - 10

7.4.1 Attributes
wear of type octet, readonly
must return transmission clutch wear (Unit: percentage, 0%: no wear, 100%: completely worn)

10.3.1 Attributes
rain of type unsigned byte, readonly
must return the amount of rain detected by the rain sensor. level of rain intensity (0: No Rain, 10:Heaviest Rain)

tripzero, 2015-05-20:
correct.  unsigned byte is a typo.  octet is probably most appropriate for each of these examples since we do not expect signed values.

acrofts84, 2015-05-20:
Thanks Kevron. 

On a similar topic, we also have a couple of unsigned shorts which expect signed values in 8.4.1. These should probably bytes instead.

>8.4.1 Attributes
mirrorPan of type unsigned short, , nullable
must return mirror pan position in percentage distance travelled, from left to right position (Unit: percentage, %0:center position, -100%:fully left, 100%:fully right)

mirrorTilt of type unsigned short, , nullable
must return mirror tilt position in percentage distance travelled, from downward-facing to upward-facing position (Unit: percentage, 0%:center position, -100%:fully downward, 100%:full upward)

djensen47, 2015-05-21:
- [ ] @acrofts84 to fix in a pull request

acrofts84, 2015-05-26:
Pull request <a href="https://github.com/w3c/automotive/pull/23">#23</a>.
Modified mirrorPan and mirrorTilt to byte to allow signed values.
Others defined as octet already in previous pull.

closed 2015-05-27 by acrofts84