w3c/automotive #13: [data-spec] Fuel Configuration Interface

acrofts84, 2015-05-20:
A couple of points on the Fuel Configuration Interface (5.4):

<ul>
<li>Fuel Type Array - I agree that this need to be an array, but we may also have to consider refuelPosition as an array as well, such as in the case of plug-in hybrids which will have both a refuelPosition for electricity and a petrol/diesel</li>
<li>FuelTypeEnum - There are two commercially available hydrogen fuel cell cars on the market (Hyundai Tucson FCEV and the Toyota Mirai). The Toyota is being released in the US in October. It may be worth distinguishing between battery electric vehicles (BEVs) and hydrogen fuel cell electric vehicles (FCEVs).
 </li>
</ul>

djensen47, 2015-05-21:
Comment: the SAE standards break this down with more detail.

- [ ] Look at all the possibilities and see what is missing
- [ ] Add missing Fuel Types
- [ ] Need to resolve zone and refuel position issue.
  - If you have 2 fuel positions, how can you determine which type of fuel is used in which position?

paulboyes, 2015-05-26:
- [ ] Adam Crofts, Greg Brannon, Paul Boyes, and Dave Jensen to bring suggestions/cases to group
- [ ] Kaz suggests potentially using WG Note to refer to for extensibility (See EmotionML for example)

acrofts84, 2015-06-12:
I've compiled a list of fuels and subtypes below. I've no doubt that it is by no means exhaustive, but offers a number of cases with bi-fuel options, reinforcing the idea of separate fuelling methods. The only additions to the existing enum would be Compressed Air and Hydrogen.

Current Enum:
<pre>
enum FuelTypeEnum {
    "gasoline",
    "methanol",
    "ethanol",
    "diesel",
    "lpg",
    "cng",
    "electric"
};
</pre>

My proposal for allowing more granularity in the spec is an additional FuelSubType, however I feel it would have to be extensible and probably

<pre>
enum FuelSubTypeEnum {
    "Regular Unleaded",
    "Premium Unleaded 98",
    "Leaded",
    "Regular Diesel",
    "Premium Diesel",
    "B5",
    "B20/30",
    "B100", 
    "E10", 
    "E15", 
    "E85", 
    "M15",
    "M85",
    "Methanol Fuel Cell"

};
</pre>

- Gasoline
 - Unleaded 95 RON
 - Super unleaded 98 RON 
 - Leaded Four Star 98 RON
- Diesel
 - 1D: Cold weather diesel (less viscous, lower energy density, worse mpg)
 - 2D: Standard diesel (more viscous, higher energy density, better mpg)
 - Regular diesel (Cetane 48)
 - Premium diesel (Cetane 55)
 - Biodiesel (Methyl Ester) 
    - B5: 5% Biodiesel, 95% Diesel. Requires no engine modification in a diesel engine
    - B20/30 (Cetane 50)
    - B100 (Cetane 55)
    - <a href="http://bioroute.co.uk/biodiesel.htm">More Info</a>
- Ethanol
 - E10 : 10% ethanol and 90% gasoline
 - E15 : 10%-15% ethanol and gasoline
 - E85 (FlexFuel) : 51%-83% ethanol and gasoline
 - <a href="http://www.afdc.energy.gov/fuels/ethanol_blends.html">More Info</a>
- Methanol
 - 100% methanol 
 - 85% methanol/15% gasoline blends
 - Fuel Cell Vehicles
 - <a href="http://www.afdc.energy.gov/fuels/emerging_methanol.html">More Info</a>
- Hydrogen
 - Fuel Cell Electric Vehicle
 - Internal Combustion
- Electric
 - Battery electric: full electric vehicle
 - Plug in hybrid: ICE and electric motor
 - Micro hybrid: stop/start technology. No electric propulsion
 - Mild hybrid: power assist from electric motor during acceleration, no electric only propulsion
 - Full hybrid: electric only, ICE only and both
- LPG (Liquefied Petroleum Gas - Propane, Butane, Natural Gas Mix)
 - Dedicated: Propane only
 - Bi-fuel: Fuelled separately (propane and gasoline), use the same engine
- CNG (Compressed Natural Gas - Methane)
 - Dedicated: CNG only
 - Bi-fuel: Fuelled separately (propane and gasoline/diesel), use the same engine
- Compressed Air
 - Air only
 - Hybrid with gasoline, diesel, ethanol

acrofts84, 2015-06-29:
So I have a bit of a cop out for this, but I'm struggling for an elegant solution. 

I'd like to propose using a string to define further information about the fuel type instead of using an enum. This method allows endless expansion and could be utilised when searching for specific fuels.

<pre>
fuelTypeVariant of type array of DOMString, readonly , nullable
must return the specific variant of fuel required by vehicle. If the vehicle uses 
multiple fuels, fuelTypeVariant returns an array of descriptions.
</pre>

In order to solve the problem of having multiple fuels in multiple positions we can use a dictionary. This is an associative array data type for creating objects in WebIDL (<a href="http://www.w3.org/TR/api-design/#when-to-use">link</a>). The types are not defined as nullable since dictionary entries are inherently optional.

If somebody with more WebIDL experience could verify this is a correct and appropriate use of a dictionary it would be appreciated!

<pre>
dictionary fuelOptions{
    FuelTypeEnum fuelType;
    DOMString fuelTypeVariant;
    Zone refuelPosition;
};
</pre>

The interface could then define an array of type fuelOptions allowing the fuel type, variant and position to be linked into an object.

<pre>
interface FuelConfiguration : VehicleCommonDataType {
    readonly    attribute fuelOptions[]? vehicleFuels;
};
</pre>

wonsuk73, 2015-07-07:
@acrofts84, Thanks a lot for good proposal! For WebIDL, I will check :)
@QingAn, @aShinjiroUrata and @paulboyes Do you have any feedback for this item?

MelcoGot, 2015-07-07:
Hi. So one question from a users/developer point of view we would have

```javascript
fuelOptions my_oem_of_some_spectic_car_fuel.
FuelTypeEnum ftype = my_oem_of_some_spectic_car_fuel[index].fuelType
DOMString subtype_s = my_oem_of_some_spectic_car_fuel[index].fuelTypeVariant
...
```

or do we have 

```javascript
fuelOptions my_oem_of_some_spectic_car_fuel.
FuelTypeEnum ftype = my_oem_of_some_spectic_car_fuel[index].fuelType
DOMString[] subtype_s = my_oem_of_some_spectic_car_fuel[index].fuelTypeVariant
...
```

First case we need to parse the string and in the second we get an array of DOMStrings representing the actual different sub types.  I think that getting an array of subtypes is prefered as stated above but I am not clear if that is the intended implementation. I think that the spec looks good!!!

acrofts84, 2015-07-07:
Having read this back I realise why this is unclear, my apologies. I have removed "If the vehicle uses 
multiple fuels, fuelTypeVariant returns an array of descriptions." when submitting the pull request. This will be a single string describing the fuel type. The fuelOptions object is the array. 

In the case of a diesel plugin hybrid:
<pre>
fuelOptions myVehicleFuels;

myVehicleFuels[0].fuelType = 'diesel'
myVehicleFuels[0].fuelTypeVariant = 'Premium Diesel'
myVehicleFuels[0].refuelPosition= ['rear', 'left'];

myVehicleFuels[1].fuelType = 'electric'
myVehicleFuels[1].fuelTypeVariant = '230VAC Single Phase'
myVehicleFuels[1].refuelPosition= ['front', 'middle'];
</pre>

Or for E85 (FlexFuel):

<pre>
fuelOptions myVehicleFuels;

myVehicleFuels[0].fuelType = 'gasoline'
myVehicleFuels[0].fuelTypeVariant = 'E85'
myVehicleFuels[0].refuelPosition= ['rear', 'left'];

myVehicleFuels[1].fuelType = 'ethanol'
myVehicleFuels[1].fuelTypeVariant = 'E85'
myVehicleFuels[1].refuelPosition= ['rear', 'left'];
</pre>

wonsuk73, 2015-08-04:
@acrofts84 For WebIDL description for FuelConfiguration using dictionary, according to [Web API Design Cookbook](http://www.w3.org/TR/api-design/#attribute-constant-or-exception-fields), it is required to be changed like below.

Your proposal
```javascript
interface FuelConfiguration : VehicleCommonDataType {
    readonly    attribute fuelOptions[]? vehicleFuels;
};
```

Updated one based on  [Web API Design Cookbook](http://www.w3.org/TR/api-design/#attribute-constant-or-exception-fields)
```javascript
interface FuelConfiguration : VehicleCommonDataType {
    readonly    attribute any vehicleFuels;
};

// with vehicleFuels is of type vehicleFuels dictionary as defined in [REFERENCE] in the spec.
```

In addition, regarding [chapter 3.1.2. of Web API Design Cookbook](http://www.w3.org/TR/api-design/#how-to-use), sequence type should be used rather than array type for vehicleFuels.

QingAn, 2015-08-06:
@acrofts84 In your case of a diesel plugin hybrid:
<pre>
fuelOptions myVehicleFuels;

myVehicleFuels[0].fuelType = 'diesel'
myVehicleFuels[0].fuelTypeVariant = 'Premium Diesel'
myVehicleFuels[0].refuelPosition= ['rear', 'left'];

myVehicleFuels[1].fuelType = 'electric'
myVehicleFuels[1].fuelTypeVariant = '230VAC Single Phase'
myVehicleFuels[1].refuelPosition= ['front', 'middle'];
</pre>
The refuelPostion uses multiple zones. Thanks to @wonsuk73 's comment, based on [chapter 3.1.2.1 of Web API Design Cookbook](http://www.w3.org/TR/api-design/#use-sequence-type), dictionary member that should carry an array of certain value, use sequence type rather than Array type. So the definition of dictionary fuelOptions is required to be changed like below.

Your proposal:
<pre>
dictionary fuelOptions{
    FuelTypeEnum fuelType;
    DOMString fuelTypeVariant;
    Zone refuelPosition;
};
</pre>

Updated one:
<pre>
dictionary fuelOptions{
    FuelTypeEnum fuelType;
    DOMString fuelTypeVariant;
    sequence&lt;Zone&gt; refuelPosition;
};
</pre>

QingAn, 2015-08-06:
@acrofts84 @wonsuk73 

I agree with @wonsuk73 's proposal that dictionaries must not be used as the type of attribute, based on  [chapter 3.1.2.4 of Web API Design Cookbook](http://www.w3.org/TR/api-design/#attribute-constant-or-exception-fields).

acrofts84, 2015-08-10:
Thank you both for your advice. I should have seen "common mistakes" and had alarm bells ringing!

I have updated pull request #47 to reflect this.

closed 2015-09-15 by acrofts84