w3c/automotive #14: [data-spec] Vehicle Speed Units - Suggestion

acrofts84, 2015-05-20:
The units of the vehicle speed vary within the spec. The average speed from the trip clock, the top speed limit and the cruise control speed are in kilometers per hour whereas the vehicle and wheel speeds are given as meters per hour.

I can see why it has been done this way for usability, but I feel consistency may be beneficial. This then gives an argument for the use of SI units. I think this is a point that has been debated before, although regardless of whether SI units or not I would argue units should be consistent for the same measurements.

>6.2.1 Attributes
speed of type unsigned short, readonly
must return vehicle speed (Unit: meters per hour)

>6.3.1 Attributes
speed of type unsigned short, readonly
must return wheel speed (Unit: meters per hour)

>6.9.1 Attributes
averageSpeed of type unsigned short, readonly , nullable
must return average speed based on trip meter (Unit: kilometers per hour)

> 6.11.1 Attributes
speed of type unsigned short, readonly
must return target Cruise Control speed in kilometers per hour (Unit: kilometers per hour)

>9.5.1 Attributes
speed of type unsigned short, readonly
must return vehicle top speed limit (Unit: kilometers per hour)

djensen47, 2015-05-21:
- [ ] Get comments on which units should be used.
  - Argument for meters per hour is that it is the small unit available to avoid floats.
- [ ] Editors: update the doc to make units consistent.

paulboyes, 2015-05-26:
- [ ] Note in spec why certain units were chosen (such as meter per hour versus kilometer per hour)

djensen47, 2015-05-26:
As a point of reference, it looks like NASA JPL uses m/s for speed:

http://www2.jpl.nasa.gov/basics/units.php

closed 2015-07-13 by acrofts84