w3c/automotive #15: [data-spec] WiperStatus Interface Superfluous?

acrofts84, 2015-05-20:
The WiperStatus interface offers the speed interval of the wipers 0 - 10:

>10.4.1 Attributes
wiperSpeed of type unsigned byte, readonly
must return current speed interval of wiping windshield (0: off, 1: Slowest, 10: Fastest )

The WiperSetting interface offers the speed setting of the wipers off to fastest and auto mode:

>10.5.1 Attributes
wiperControl of type WiperControl,
must return current setting of the front wiper controller. It can be used to send user's request for changing setting.

I would suggest a boolean value for auto mode instead of holding it within the WiperControl enum and then just a single indication of WiperSpeed. <i>It can be used to send user's request for changing setting</i> - if this is the reason for the separate WiperSetting interface then WiperSetting and the WiperStatus should adhere to the same enum.

aShinjiroUrata, 2015-05-21:
It looks WiperStatus and WiperSetting have been already integrated into one at below commit.
https://github.com/w3c/automotive-bg/commit/a52783ed47a8415cd319b5b8276db61146d359f6

If you are checking below url which is written in BG web site, bit confusng but it is not the lates one. 
http://www.w3.org/2014/automotive/data_spec.html

Latest version is like this.
<pre>
10.4 WiperStatus Interface

enum WiperControl {
    "off",
    "once",
    "slowest",
    "slow",
    "middle",
    "fast",
    "fastest",
    "auto"
};

interface WiperStatus : VehicleCommonDataType {
    readonly    attribute WiperControl wiperSpeed;
                attribute WiperControl wiperSetting;
    readonly    attribute Zone?        zone;
};

10.4.1 Attributes

wiperSetting of type WiperControl,
  must return current setting of the front wiper controller. It can be used to send user's request for changing setting.

wiperSpeed of type WiperControl, readonly
  must return current speed interval of wiping windshield

zone of type Zone, readonly , nullable
  must return Zone for requested attribute
</pre>

acrofts84, 2015-05-21:
Ok, great, thank you!

closed 2015-05-21 by acrofts84