w3c/automotive #17: [data-spec] Steering Wheel Left - Ambiguous

acrofts84, 2015-05-21:
The steeringWheelLeft attribute name feels very ambiguous to me. Would this be better off as a Zone? Or a less ambiguous term e.g. leftHandDrive or driverSideLeft. Although I don't know how well these translate.

<pre>5.7.1 Attributes
steeringWheelLeft of type boolean, readonly , nullable
must return true if steering wheel is on left side of vehicle
</pre>

djensen47, 2015-05-22:
- [ ] Ask previous editors for their comments on this.

paulboyes, 2015-05-26:
- How to handle cars with no steering wheel?
- How to handle single seat car?
- Center steer wheel?
- Other interfaces?  Such as head unit location?  What similar interfaces are we missing?

tripzero, 2015-05-26:
You could make it steeringWheelOrientation with just a zone attribute.

closed 2015-07-13 by acrofts84