w3c/automotive #18: [data-spec] Power Modes - Cranking

acrofts84, 2015-05-21:
At the moment we only have 4 power modes in the spec. Engine cranking (start ignition) would be a useful addition to this enum.

<pre>enum VehiclePowerMode {
    "off",
    "accessory1",
    "accessory2",
    "running"
};</pre>

E.g. here are BMWs powermodes:
KL 0 = OFF (ignition off position)
KL R = ignition position 1 ("run" - some electronics & modules are powered up)
KL 15 = ignition position 2 ("accessory" - all electronics & modules are powered)
KL 30 = ignition position 3 (where the ignition defaults after starting the engine - running)
KL 50 = ignition start position (engine cranking)

djensen47, 2015-05-22:
Need a discussion with the group. 

Either go with the lowest common denominator or support variability or a combination of the two.

paulboyes, 2015-05-26:
- [ ] Adam Crofts to make pull request to add "cranking".  Differentiate between acc1 and acc2.

acrofts84, 2015-05-27:
Pull request <a href="https://github.com/w3c/automotive/pull/30">#30</a>

closed 2015-07-13 by acrofts84