w3c/automotive #22: [data-spec, vehicle-spec] Remove etc from specification

peterMelco, 2015-05-26:
From vehicle_spec.html : Abstract, 12
From data_spec.html : 6,6.26,6.27,12.1.2,12.1.3

A specification should contain etc...either we need to define etc or we need to remove.

paulboyes, 2015-05-26:
- [ ] Peter Winzell to go through the cases where "etc" is used and propose alternatives as appropriate.

peterMelco, 2015-06-16:
target to deliver a review of etc by 26/6-15

MelcoGot, 2015-07-07:
I have forked and made some changes.  I will now add a pull request for this and  If this is the wrong procedure please let me know of the correct way.  The review of this should be done of some one who is a native speaker of English. So, we avoid any limitations that the removal of etc brings.

acrofts84, 2015-07-13:
Pull request <a href="https://github.com/w3c/automotive/pull/48">#48</a>

wonsuk73, 2015-08-04:
Pull request #48 made conflict to be merged. So it has been done via Pull request #57 and #58

closed 2015-08-04 by wonsuk73