w3c/automotive #32: 7.10 - Diagnostic Interface

gbrannonaaa, 2015-05-27:
For Diagnostic interface in 7.10, some additional data elements that could be considered

- diagnostic trouble codes - array
	- including 213 EPA mandated codes and OEM-specific, as available

wonsuk73, 2015-06-19:
@QingAn @acrofts84 @aShinjiroUrata Could you share your view on this issue? I would like to have more feedback for this item.

aShinjiroUrata, 2015-06-22:
If this is interface for DTC(diagnostic trouble code) of OBD2, maybe better to have this interface, I suppose.

gbrannonaaa, 2015-06-22:
Our view if that is seems very odd to omit this very important piece of information for the owner of the vehicle.  Someone using the API could utilize the DTC to inform the consumer about issues that are going on with their car.  The business group limited spec to only include information about the MIL lamp.  This is similar to the doctor telling you that you are sick, that you have been for a while, but not telling you why or what part of your body is the problem.  The spec should not be limited to only malfunctions that turn on the MIL, but any DTC.  The API user can determine what they wish to do with the data.

wonsuk73, 2015-06-26:
@aShinjiroUrata Thanks! @gbrannonaaa Thanks for additional info!. I have couple of questions!

1) When does this TroubleCode interface be used? We also have VehicleInterfaceError Interface in Vehicle Information Access API. What is different between them in the aspect of a role?

2)  In addition, we have simple example code [1] in the intro section of API spec. So could you make some example code how we could use TroubleCode interface?

[1] http://rawgit.com/w3c/automotive/master/vehicle_data/vehicle_spec.html#introduction

acrofts84, 2015-06-26:
Some cars do have methods for accessing their DTCs but they seem fairly obfuscated and I think there is an important reason for this. Exposing this data to a wider audience is more likely to increase customer dissatisfaction. 
A DTC is meant to be logged on the vehicle if the driver <i>could</i> notice an issue. If it is important, then a warning light will come on. If they don't notice, then there is no issue as far as they are concerned. If they do notice, there will be a physical symptom which they can investigate.
If you have use cases where we can equally obfuscate this information from a casual driver and then aid the more savvy drivers then I'm happy to listen. Although in the age where cars are essentially computers with wheels there are probably going to be more potential DTCs, and less that the savvy driver can do about them without the appropriate software.

gbrannonaaa, 2015-06-29:
My position is that this data should be available to the user of the API.  The DTCs (generic and OEM specific)  can be screened by the API user for what may be appropriate for their individual use case.  Hundreds of use cases can be established for this information.  Going to the trouble of calling out that the MIL light is on, but proving no information about why it is on seems to miss the point of having an interface with the vehicle altogether.

QingAn, 2015-06-30:
If this data is for the API user, I suppose better to add this.

Again, as @wonsuk73 suggested, how could the diagnostic trouble codes be used? Any example?

gbrannonaaa, 2015-07-06:
The diagnostic trouble codes could be used in a variety of way.  Perhaps this goes backs to the need to develop additional use cases, but I was respecting Paul's earlier advice that spending time to build all the use cases is not as productive as actually working on the API.  One user of the API could be a non-technical customer, to Adam's point, that wouldn't want to know what was going on with any trouble codes in their car until they needed to do something about it.  Another user could be a third party that, with the customer's consent, monitors the vehicle's systems and based on trends alerts the customer when a fault is beginning to occur based on historical data and codes coming from the vehicle that are not otherwise apparent to the customer at all (MIL is not on).  A third use could be for the repair industry to interface with the vehicle in a new way to receive vehicle data.  The list goes on and on.  Ignoring this data seems to miss the intention of the open architecture that will allow developers to build applications that will benefit consumers in this way.

gbrannonaaa, 2015-07-06:
One use case from GM

http://www.digitaltrends.com/cars/chevrolet-proactive-service-alerts-pictures-news/

wonsuk73, 2015-07-07:
@gbrannonaaa, Thanks for useful info!

closed 2015-07-16 by acrofts84