w3c/automotive #33: DriveMode and DrivingMode

aShinjiroUrata, 2015-05-30:
<pre>
6.26 DrivingMode Interface
8.6 DriveMode Interface
</pre>
I felt similar name and may be confusing. 
If `DrivingMode` is not commonly used name, how about `DrivingStatus` or some other name?
(I'm not sure about suitable naming.)

aShinjiroUrata, 2015-06-18:
It looks the name '6.26 DrivingMode' is selected to align with '6.27 NightMode', '6.28 StartStopMode' and so I guess this naming is reasonable from this view point.

On the other hand, '8.6 DriveMode' looks natural name.

Since I don't come up with better alternative for 'DrivingMode', I'd like to keep this as is if other people don't have other opinion.

acrofts84, 2015-06-18:
I agree that they are very similar. Would 'VehicleInMotion' or 'MotionStatus' be a more appropriate name instead of 'DrivingMode'?

aShinjiroUrata, 2015-06-19:
Thanks for the input.
My concern is just the names were too similar and so, both `VehicleInMotion`, `MotionStatus` are OK to me. 
For me, I like shorter one. 
If there is no other opinion, I will create pull request with `MotionStatus`.

aShinjiroUrata, 2015-07-09:
As pull request #46 "renamed DrivingMode to DrivingStatus" already merged, I close this issue.

closed 2015-07-09 by aShinjiroUrata