w3c/automotive #40: [vehicle_spec] #7 ZoneInterface - add enumerations to ZonePosition for vertical axis

drkevg, 2015-06-04:
In Vehicle Spec. section 7, suggest augmenting the existing ZonePosition values which currently define longitudinal (front/back) and horizontal (left/right) position with additional values that define vertical position in vehicle. 

Suggested new values would be:

top: top layer of vehicle in vertical-axis
middle: middle layer of vehicle in vertical-axis
bottom: bottom layer of vehicle in vertical-axis

This would allow us to distinguish the ZonePosition for cameras fitted 'top', 'rear', 'centre' from one fitted 'bottom', 'rear', 'centre'

Also suggest following the same naming system used by CSS for vertical and horizontal alignment. 

So in the vertical-axis : top, middle, bottom
In the horizontal-axis  : left, center, right
In the longitudinal-axis: front, central, rear

Suggested enumeration Descriptions would then be:

top: top layer of vehicle in vertical-axis
middle: middle layer of vehicle in vertical-axis
bottom: bottom layer of vehicle in vertical-axis
left: left area of vehicle in horizontal-axis
center: center area of vehicle in horizontal-axis
right: right area of vehicle in horizontal-axis
front: front of vehicle in longitudinal-axis
central: central area of vehicle in longitudinal-axis
rear: rear area of vehicle in longitudinal axis

Some example Zone values could then be:

zone.ZonePosition[0] = 'left'  				// denoting the whole left plane of the vehicle
or
zone.ZonePosition[0] = 'left', 'front' 	        	// denoting the left, front area of the vehicle
or
zone.ZonePosition[0] = 'left', 'front', 'bottom'    // denoting the left, front, bottom of the vehicle
or
zone.ZonePosition[0] = 'front', 'left' 	        	// denoting the front, left and
zone.ZonePosition[1] = 'back', 'right' 	        	// back, right areas of the vehicle
or
zone.ZonePosition[0] = 'front','top','center'       	// denoting front, top, center and 
zone.ZonePosition[1] = 'back', 'bottom', center'	// back, bottom, center areas of the vehicle

 
Combined with the recent proposal to ensure that Zone contains an array of ZonePositions, suggest that this would enable the existing approach to cover all three physical axis in the car

QingAn, 2015-06-05:
+1
@drkevg Could you make a pull request to modify this?

aShinjiroUrata, 2015-06-05:
Hi, 
I agree to the proposal.

By the way, just technical point.
I was thinking, to read the definition literally, zone spcification would be like this.
This means one `Zone` specifies one position by combination of 1,2 or 3 `ZonePosition`. 
```javascript
var zone = new Zone();
zone.value[0] = 'left';
zone.value[1] = 'front';
zone.value[2] = 'bottom';
```
As written
> zone.ZonePosition[0] = 'left' // denoting the whole left plane of the vehicle
> zone.ZonePosition[0] = 'left', 'front' // denoting the left, front area of the vehicle
> zone.ZonePosition[0] = 'left', 'front', 'bottom' // denoting the left, front, bottom of the vehicle

If I want to specify plural items in an element of array, it would be like 
```javascript
var zone = new Zone();
zone.value[0] = ['left', `front`, `bottom`];
zone.value[1] = [`left`, 'rear'];
```

Then I suppose, Zone definition would be like this. (This is ok to me.)
This means one `Zone` value can hold plural position information.
```javascript
interface Zone {
    attribute ZonePosition[][] value;          //before: attribute ZonePosition[] value;
    readonly    attribute Zone           driver;
    boolean equals (Zone zone);
    boolean contains (Zone zone);
};
```

On the other hand, when I look at this definition,  
```javascript
interface VehicleInterface {
    Promise get (optional Zone zone);
    readonly    attribute Zone[] zones;
};
```
I feel implication that one `Zone` has one position information and it looks inconsistent.
Then this should be like below?
```javascript
interface VehicleInterface {
    Promise get (optional Zone zone);
    readonly    attribute Zone zones;  // because one Zone can hold multiple position. 
};
```

wonsuk73, 2015-06-16:
Agreed with proposals from @drkevg

@aShinjiroUrata  Regarding plural items support in Zone interface, do you have use cases for this? I need better understanding for this! Thanks!

@QingAn @tripzero @acrofts84 any comment on this?

aShinjiroUrata, 2015-06-16:
Hi Wonsuk-san,
I'm not proposing something but I just wanted to clarify the use of Zone.

To read the spec literally, 
http://rawgit.com/w3c/automotive/master/vehicle_data/vehicle_spec.html#zone-interface
I thought the use of zone was like this. I mean one ZonePosition enum in one value[] item.
```javascript
var zone = new Zone();
zone.value[0] = 'left';
zone.value[1] = 'front';
zone.value[2] = 'bottom';
```

What @drkevg wrote is slight different, in which plural ZonePosition enums are set in one value[].
```javascript
// I replaced '.ZonePositon' with '.value' since 'value' is right attribute name.
zone.value[0] = 'left', 'front', 'bottom' // denoting the left, front, bottom of the vehicle
or
zone.value[0] = 'front', 'left' // denoting the front, left and
zone.value[1] = 'back', 'right' // back, right areas of the vehicle
or
zone.value[0] = 'front','top','center' // denoting front, top, center and 
zone.value[1] = 'back', 'bottom', center' // back, bottom, center areas of the vehicle
``` 

I just wanted to say, if to allow the @drkevg's way, I suppose, Zone Interface definition have to be changed.

Personally I think @drkevg's way is more convenient because, in the former way,
 it is difficult to specify front-left and rear-right in one Zone.
```javascript
// This looks, front-left, front-right, rear-left, rear-right are specified.
var zone = new Zone();
zone.value[0] = 'front';
zone.value[1] = 'left';
zone.value[2] = 'rear';
zone.value[3] = 'right';
```
```javascript
// Need two Zones to specify front-left and rear-right
var zone1 = new Zone();
zone1.value[0] = 'front';
zone1.value[1] = 'left';

var zone2 = new Zone();
zone2.value[2] = 'rear';
zone2.value[3] = 'right';
```

By @drkevg's way, possible to specify front-left, rear-right in one Zone.
```javascript
var zone = new Zone();
zone.value[0] = ['front', `left`];
zone.value[1] = [`rear`, 'right'];
```

I hope this explains my question enough.

drkevg, 2015-07-07:
Apologies for any confusion introduced by incorrectly specifying ZonePosition (the property type) instead of 'value' (the property name) in the examples I gave.

Re: the request for a use case, an example use case would be specifying camera positions. 

Say a vehicle had two front facing cameras positioned at the front, bottom, left and front, bottom, right corners of the vehicle and one rear facing camera positioned at the rear of the vehicle, in the top, center  (e.g. connected to inside of roof pointing through rear window)

We would have three cameras, each positioned in a different vehicle zone: 

front, bottom, left
front, bottom, right

and 

rear, top, center

Suggest that it would be possible to define the zone for the first camera as follows:

camera[0].zone.value = ['front','bottom','left'];

Which would be equivalent to:

/* Define Zone for Camera 1 */
camera[0].zone.value[0] = 'front';
camera[0].zone.value[1] = 'bottom';
camera[0].zone.value[2] = 'left';

Could then define the zones for the other two cameras using either:

camera[1].zone.value = ['front','bottom','right'];   /* Camera 2 */
camera[2].zone.value = ['rear','top','center'];      /* Camera 3 */

OR

/* Define Zone for Camera 2 */
camera[1].zone.value[0] = 'front';
camera[1].zone.value[1] = 'bottom';
camera[1].zone.value[2] = 'right';

/* Define Zone for Camera 3 */
camera[2].zone.value[0] = 'rear';
camera[2].zone.value[1] = 'top';
camera[2].zone.value[2] = 'center';

aShinjiroUrata, 2015-07-09:
Dear Kevin-san,
Thank you very much for the clarification.
My question was solved.

closed 2015-09-15 by acrofts84