w3c/automotive #44: Refactor avaiableFor methods

j-hashi, 2015-06-26:
1) Parameter of availableFor() doesn't need.  Let's remove it.
```js
    Availability availableForRetrieval ();
    Availability availableForSubscription ();
    Availability availableForSetting ();
```

2) Methods name should be consistent to the related methods. Let's change the name to:
```js
    Availability availableForGetting ();
    Availability availableForSubscription();
    Availability availableForSetting ();
```

3) They doesn't need to be methods. Let's change them to attributes.
```js
    attribute Availability gettingAvailability;
    attribute Availability subscriptionAvailability;
    attribute Availability settingAvailability;
```

Then the example 5 will be shorten to
```js
if( ( var a = vehicle.vehicleSpeed.getttingAvailability ) === "available" )
{
  // we can use it.
}
else
{
  // tell us why:
  console.log(a);
}
```

wonsuk73, 2015-07-29:
@j-hashi I couldn't look for "availableFor()" parameter in the spec. So could you make clear the meaning of "1) Parameter of availableFor() doesn't need."? Agreed with "2)". For "3)", it might be okay. 

In general, I am curious why we need three method or attribute for this. one method with one more parameter or one attribute with different value can do the same.

For one method example,
 var speedAvailability = vehicle.vehicleSpeed.availability("speed","GET"); // GET&SET, SUBSCRIPTION, etc

For one method example,
 if( ( var a = vehicle.vehicleSpeed.getttingAvailability ) === "GET" ) // GET&SET, GET&SUBSCRIPTION, etc

@QingAn @acrofts84 @djensen47  Do you have any opinion on this?

j-hashi, 2015-07-29:
@wonsuk73 
I meant three availableFor****() methods:
```JS
Availability availableForRetrieval ();
Availability availableForSubscription ();
Availability availableForSetting ();
```

wonsuk73, 2015-07-29:
@j-hashi I understood that "availableFor****()" methods don't need a parameter. I agreed with that. Thanks!

tripzero, 2015-07-29:
I agree that they can be attributes instead of methods.

wonsuk73, 2015-08-04:
@tripzero I would like to check your position for my proposal which is to merge three attributes to one. What do you think?

wonsuk73, 2016-11-08:
This issue is related with the vehicle information access API spec which is not developed in the group anymore.

closed 2016-11-08 by wonsuk73