w3c/automotive #55: [vehicle_spec] recursive definition in Zone not problem?

aShinjiroUrata, 2015-07-29:
```javascript
interface Zone {
                attribute ZonePosition[] value;
    readonly    attribute Zone           driver;
    boolean equals (Zone zone);
    boolean contains (Zone zone);
};
```
`interface Zone` uses `Zone` itself inside the definition.
To think straight forward, constructor of `Zone` need to construct another `Zone` inside and it is recursive. I suppose this is problematic.
To avoid this recursion, one idea is to use `getDriverPosition()` method.

But another thing I think is, if we need driver position inside `Zone` definition or not.
It looks we can get driver position from `interface steeringwheelConfigutation`.
So I guess it would be no problem to remove driver position from `Zone` definition. 

```javascript
interface Zone {
                attribute ZonePosition[] value;
    boolean equals (Zone zone);
    boolean contains (Zone zone);
};
```

tripzero, 2015-07-29:
Whether or not this becomes recursive is an implementation detail.  I got
it to work avoiding infinite recursion by, iirc, overriding the getter in
the "driver" attribute and making it not enumerable (which iirc is the
default).

aShinjiroUrata, 2015-07-30:
@tripzero thanks for the comment.
I suppose you mean this.
```javascript
window.Zone = function(zone) {
    this.value = zone ? zone : [];

    Object.defineProperty(this, 'driver',
                          { enumerable: false, get: function() {
                              return new Zone(['Front', 'Left']);
                          } });
};
```
https://raw.githubusercontent.com/otcshare/automotive-message-broker/master/xwalk/vehicle_api.js

OK. I get it.
We are thinking about C++ layer implementation and maybe there is some workaround.

Then how about this point?
>But another thing I think is, if we need driver position inside Zone definition or not.
>It looks we can get driver position from interface steeringwheelConfigutation.
>So I guess it would be no problem to remove driver position from Zone definition.

From a point of view, driver position is suitable to be handled by Configuration and Ideintification interface and enough by having steeringWheelConfigutation interface.
Is there a special reason to have it inside Zone interface?

tripzero, 2015-07-31:
There is a valid point about the redundancy.  There should only be one way
to avoid confusion.

aShinjiroUrata, 2015-08-04:
I understand.
Personally, still I think, driver position can be  located in Configuration and Identification interface.

closed 2015-08-04 by aShinjiroUrata