w3c/automotive #61: [vehicle_spec] Inconsistent data type and data unit (From IBM Steven Atkin)

QingAn, 2015-08-26:
1. Inconsistent data type
 - Inconsistent data type for levels
   7.6.1 WasherFluid Interface
http://www.w3.org/TR/2015/WD-vehicle-data-20150616/#attributes-43
Across the specification levels use the octet data type while in this case the level uses an unsigned short. The same data type should be used across the specification. 
For example, in 6.19 EngineCoolant http://www.w3.org/TR/2015/WD-vehicle-data-20150616/#idl-def-EngineCoolant  The level is an octet.

 - Inconsistent data type for temperature
   10.2.1 Temperature Interface 
http://www.w3.org/TR/2015/WD-vehicle-data-20150616/#attributes-66
In this interface temperatures use the float data type, while other interfaces in the specification use the long data type. Does this interface specifically require the use of float or should this be made to be consistent with the other uses of temperature?
For example, in 6.17 EngineOil Interface
http://www.w3.org/TR/2015/WD-vehicle-data-20150616/#attributes-25  The temperature is a long
For example, in 6.19.1 Engine Coolant 
http://www.w3.org/TR/2015/WD-vehicle-data-20150616/#attributes-27 The temperature is a short while in section 6.17.1 temperature is a long. 
In section 7.3 TransmissionOil Interface
http://www.w3.org/TR/2015/WD-vehicle-data-20150616/#attributes-40 The temperature is a byte.

 -  Inconsistent use of data type for tilts and pans
    8.4 Mirror Interface
http://www.w3.org/TR/2015/WD-vehicle-data-20150616/#idl-def-Mirror
Throughout the specification tilts and pans use different data types. In this particular case the data type is a byte while in other sections tilt is an octet or an unsigned short.
See section 10.6 Sunroof Interface http://www.w3.org/TR/2015/WD-vehicle-data-20150616/#idl-def-Sunroof
See section 5.7 Steering Wheel Configuration http://www.w3.org/TR/2015/WD-vehicle-data-20150616/#idl-def-SteeringWheelConfiguration

2. Inconsistent data unit
 - Should units of speed vary in the specification
   6.2 Vehicle Speed Interface
http://www.w3.org/TR/2015/WD-vehicle-data-20150616/#idl-def-VehicleSpeed
In the specification the definition of speed is meters per hour in some interfaces and kilometers per hour in other interfaces.
See 6.9 Trip Interface http://www.w3.org/TR/2015/WD-vehicle-data-20150616/#idl-def-Trip
Should the definition of speed be consistent across all interfaces?

acrofts84, 2015-09-02:
<b>Inconsistent data type for levels</b>

The octet type is an unsigned integer type that has values in the range [0, 255].
The unsigned short type is an unsigned integer type that has values in the range [0, 65535].
The byte type is a signed integer type that has values in the range [−128, 127].

Therefore unsigned percentages should be octet, and signed percentages byte.

At the moment in the spec there are 16 as octet and 8 as unsigned short. I will change the unsigned shorts to octets.
___________________________________________________________________

<b>Inconsistent data type for temperature</b>
I think float is the most appropriate value for this. It will be accurate to a decimal place therefore floating point calculations should not degrade precision considering the precision of the sensors. None of these values will ever require the range of a long.
__________________________________________________________________

<b>Inconsistent use of data type for tilts and pans</b>
This is because all other data types are unsigned whereas these values are signed, therefore I don't think any change is appropriate.
___________________________________________________________________

<b>Inconsistent data unit for speed</b>
This has already been raised and fixed for all speeds to be meters per hour, see <a href="https://github.com/w3c/automotive/issues/14">Issue 14</a>

wonsuk73, 2015-09-11:
I agreed all of summary from @acrofts84 !

closed 2015-09-15 by acrofts84