w3c/automotive #62: [vehicle_spec] Use BCP47 instead of ISO 639-1 for identifying languages (From IBM Steven Atkin)

QingAn, 2015-08-26:
8.2 Language Configuration Interface
http://www.w3.org/TR/2015/WD-vehicle-data-20150616/#attributes-49
BCP47 should be used to identify languages instead of ISO 639-1.
See https://tools.ietf.org/html/bcp47
One of the problems with using ISO 639-1 2 letter codes is that it does not adequately address the handling of Chinese. For example, in ISO 639-1 Chinese is represented as zh. This code cannot indicate whether the writing system is Simplified Chinese or Traditional Chinese. The BCP47 standard clearly indicates which writing system is in use. In this case if Simplified Chinese was being used the code would be zh-Hans and for Traditional Chinese it would be zh-Hant.

10 Vehicle Interface Error
http://www.w3.org/TR/2015/WD-vehicle-information-api-20150616/#idl-def-VehicleInterfaceError
Which languages may be used in the error message? If languages other than English are permitted then you may want to add an attribute to the interface to clearly indicate the language of the message. 
You may want to consider modifying this interface to permit an array of messages tagged with their language so that the interface could return messages in more than one language. This would enable an end-user to select from a list of possible languages to see messages in.
If you add a language tag then you should follow the BCP47 recommendations https://tools.ietf.org/html/bcp47
Do not use the ISO639-1 2 letter codes for languages.

QingAn, 2015-08-26:
I agree that we should use BCP47 to identify languages instead of ISO 639-1, as explained above.

@paulboyes @wonsuk73 @aShinjiroUrata @gbrannonaaa @acrofts84 What's your view on this?

aShinjiroUrata, 2015-08-27:
Hi,
For `8.2 Language Configuration Interface`, I agree to this comment.

For `10 Vehicle Interface Error`, basically OK with me but one thing in mind is,
`Vehicle Interface Error` is returned from Promise when `get` `set` is rejected and I guess this error message is usually not directly shown to users.
(although, `message` desciption says **user-readable error message**)

I guess `Vehicle Interface Error` is something similar to `5.5 PositionError interface` of geolocation api. 
http://dev.w3.org/geo/api/spec-source.html#position-error

`message` of `PositionError interface` is explained like
> The message attribute must return an error message describing the details of the error encountered. 
>This attribute is primarily intended for debugging and developers should not use it directly in their application user interface. 

If `message` of `10 Vehicle Interface Error` is primarily for debug purpose as well, I guess supporting more than one language is not very much necessary and just one language is good in terms of simplicity.

acrofts84, 2015-09-02:
I also agree for the Language Configuration interface.

For the vehicle error interface I don't think it would be relevant. If the implementer uses multiple languages they would probably use a text placeholder and a database instead of hard-coded text. They would then determine the actual text to send from the language configuration interface. Let me know if I am missing the point.

wonsuk73, 2015-09-30:
@QingAn, Hi. When I took a look at above discussion, It seems we need to change ISO 639-1 with BCP 47 for "8.2 Language Configuration Interface". It seems it didn't has done yet. Am I right? 
But this issue had been closed already so I am curious.

wonsuk73, 2015-09-30:
@aShinjiroUrata @acrofts84 @QingAn 

For 10 Vehicle Interface Error, I am in the same position with you guys. But I think to make clear for readers, it's required to add additional sentense like below text (from `message` of `PositionError interface` of Geolocation API)

> The message attribute must return an error message describing the details of the error encountered. > This attribute is primarily intended for debugging and developers should not use it directly in their application user interface.

wonsuk73, 2015-09-30:
I made a pull request #69 to make clear sentence for "message" attribute. @aShinjiroUrata @acrofts84  @QingAn, Any comment on this?

aShinjiroUrata, 2015-10-01:
@wonsuk73, I agree to your opinion.

Thanks for creating the pull request.
It looks the sentence is terminated in the middle.

I think better to add 
> should not use it directly in their application user interface.

wonsuk73, 2015-10-01:
@aShinjiroUrata Thanks for quick comment! Okay! I will update the pull request with your comment!

closed 2015-10-02 by wonsuk73