w3c/automotive #75: API Test plan

peterMelco, 2015-12-10:
**Vehicle Information Access API - Test plan.**

**How to test**: w3c.github.io/testing-how-to
Look for conformance requirements in the spec:
 "In the simplest case, a conformance requirement is any single statement that uses RFC 2119 language (MUST, MUST NOT, SHOULD, SHOULD NOT, MAY), that specifies required behavior for a particular conformance class defined elsewhere in the spec (e.g., a Web browser or other user agent).”

Ex 1:
From the spec“ Implementations that use ECMAScript to implement the APIs defined in this specification **must** implement them in a manner consistent with the ECMAScript Bindings defined in the Web IDL specification [WEBIDL], as this specification uses that specification and terminology"

```javascript
<!DOCTYPE html>
<html>
<head>
<meta charset="utf-8">
<title>Web Storage IDL tests</title>
<link rel="author" title="W3C" href="http://www.w3.org/" />
<link rel="help" href="http://www.w3.org/TR/webstorage/#storage"/>
<script src="/resources/testharness.js"></script>
<script src="/resources/testharnessreport.js"></script>
<script src="/resources/WebIDLParser.js"></script>
<script src="/resources/idlharness.js"></script>
</head>
<body>
<h1>Vehicl Access API  IDL tests</h1>
<div id="log"></div>
<pre id='untested_idl' style='display:none'>
[PrimaryGlobal]
interface Window {
};
interface Event {
};
interface EventInit {
};
</pre>
<pre id='idl'>
interface Vehicle {
 readonly attribute VehicleSignalInterface vehicleSpeed;
 readonly attribute VehicleSignalInterface wheelSpeed;
 …	
		};
...
<script>
(function() {
	var idl_array = new IdlArray();
	idl_array.add_untested_idls(document.getElementById("untested_idl").textContent);
	idl_array.add_idls(document.getElementById("idl").textContent);
	idl_array.add_objects({Vehicle: ["navigator.vehicle"]});
	idl_array.test();
})();
</script>
...
</html>
```

Ex 2:
From spec: "driver of type Zone, readonly **mus**t return physical zone for logical driver"
So, test that the driver is either front right or front left (any of the valid logical driver positions…)

```javascript
...
 Assert_equal(driver.value,leftfrontzone) || Assert_equal(driver.value,rightfrontzone)
…
```

**Suggestion to follow the *Test the Web Forward*: http://w3c.github.io/testing-how-to/#(11)**

Why follow Test the Web Forward?

- They know how to test the Web
- They want the Web to work, no exceptions
- A growing number of browser vendors are running the TTWF tests
- They have a lot of test infrastructure ready to use
- Your lifespan is too short to reinvent it
- They have happy and cool dinosaurs on their landing page

**Next step** 
http://testthewebforward.org/docs/  
So, basically fork the test repository and submit our tests Vehicle/xxx.hmtl using pull requests. 

**Who will  contribute ?** 
**Actions ?**
**Next meeting ?**

peterMelco, 2016-01-19:
We need to decide what we should actually test. Should we write tests that can be used to **test an implementation of the specification** and/or should we **test the specification** ? Taking Geolocation as an example they test an implementation of the geolocation specification, and this seems to be the most reasonable thing to do. However, the IDL itself should be tested with https://w3c.github.io/testing-how-to/#(26). Now, using mockup data on the javascript/browser layer assumes that we have persistence in the IDL layer (xI.DL + x.h, x.cpp) and application logic. However, this is implementation dependant and, therefore we should not base our tests on mockup data. Typically, you would have a DataManager process that would handle this logic and the persistence. This Manager would communicate with various I/O channels on vehicle network. It is of course on this level one should use mockup data, but this is out of scope for the W3C.

peterMelco, 2016-01-19:
Mistake should not be closed

ashimura, 2016-01-19:
reopening the issue-75

ashimura, 2016-01-19:
The main purpose of getting implementation experience and generating
tests (test assertions and test suites) is to see:
- if each feature of the spec is implementable
- if there are independent (more than one) implementations for the spec
- etc.

So the purpose is not checking the quality, stability, etc., of the implemented
products but checking the spec's implementability.

Please see also the W3C Process Document at:
  https://www.w3.org/2015/Process-20150901/#implementation-experience

peterMelco, 2017-01-05:
Ome question raised from HIra-san test assertion excel sheet was whether or not to use wildcards with getVSS.  My understanding of getVSS is that it a mean for a client to ask the server what is there ? Then why would we need wildcards at all ? On the other hand getVSS could server additional pruposes and specify what is there for a specific branch and then it would make perfect sense to use wildcards. 

**ex:
getVSS. no wildcards would return the entire tree...
getVSS Vehicle.Engine.*, would return all signals available below Engine.**

I guess we should argue why we can't use a normal get request for this purpose ?

Hirabayashi-KDDI, 2017-01-05:
I would suggest that to facilitate reviews and comments,  #75 changes to a role of general reception for questions of testing , and split issues on testing in the following pieces:
#1xa: Implementation Report Plan
#1xb: Test Assertion
#1xc: Testing system and methods
#1xd: Test Results.
If it's OK,  I will make these issues of Github and upload some materials for testing by Jan 10, 2017.

peterMelco, 2017-01-09:
Ok with me

Hirabayashi-KDDI, 2017-01-10:
Based on the recent conference calls and Peter's feedback, 
I have changed and set up some issues on the GitHub as follows:

#75  - "API Test Plan"
       (https://github.com/w3c/automotive/issue/75)
        The existing issue #75 changes to a role of general reception and index for questions and      
        comments of testing.

#123 - "Implementation Report Plan" 
       (https://github.com/w3c/automotive/issue/123)
       I have placed a draft (pptx version 01) of Implementation Report Plan (IRP)
       as a new issue #123 on the GitHub.
       In a few days, web style document will be available.
       If any, please put comments on the issue.

#124 - "Test Assertion"
       (https://github.com/w3c/automotive/issue/124)
       A draft (version 01) of test assertion list will be found in a new issue #124 of the GitHub.
       If any change (editing, deletion, adding) would be proposed, please be kindly requested to add a 
      comment on #124 
     
#125 - "Testing system"
       (https://github.com/w3c/automotive/issue/125)       
       Testing architecture and test module corresponding to each item in test assertion
       will be put here.

#126 - "Test Results"
       (https://github.com/w3c/automotive/issue/126)
       This repository is provided to store the test results performed by each implementers,
       basically in forms of test assertion list denoted in #124.

peterMelco, 2017-01-10:
Hi Hira-san!
The links below are incorrect …should be issues and not issue in URL.
A also labeled the issues with testing label.
Br
Peter Winzell

(https://github.com/w3c/automotive/issue/75)

(https://github.com/w3c/automotive/issue/123)

(https://github.com/w3c/automotive/issue/124)

(https://github.com/w3c/automotive/issue/125)

(https://github.com/w3c/automotive/issue/126)

ashimura, 2017-01-10:
@peterMelco 
It seems the links in Hira-san's comment look good.
Could you elaborate what the problems are like?

Kazuyuki

Hirabayashi-KDDI, 2017-01-10:
Peter-san

Thank you very much.
I will fix all urls (issue to issues) of links in my e-mail, pptx file 
and the Github.
After that, I will resend e-mail to members.

Best regards,

T.Hirabayashi 

----- Original Message -----
Hi Hira-san!
The links below are incorrect …should be issues and not issue in URL.
A also labeled the issues with testing label.
Br
Peter Winzell

(https://github.com/w3c/automotive/issue/75)

(https://github.com/w3c/automotive/issue/123)

(https://github.com/w3c/automotive/issue/124)

(https://github.com/w3c/automotive/issue/125)

(https://github.com/w3c/automotive/issue/126)
https://github.com/w3c/automotive/issues/124>.

https://github.com/w3c/automotive/issues/75#issuecomment-271556549
　　　　　　　　　　～～～～～～～～～～～～～～～～～～
　　　　　　　　　　ＫＤＤＩ総合研究所

　　　　　　　　　　平林　立彦
　　　　　　　　　　HIRABAYASHI Tatsuhiko
　　　　　　　　　　Tel: 03-6678-1946(代表)
　　　　　　　　　　Mobile: 080-5941-4506
　　　　　　　　　　Fax: 03-6678-0339
　　　　　　　　　　E-mail: ta-hirabayashi@kddi.com
　　　　　　　　　　Address: 〒102‐8460
　　　　　　　　　　東京都千代田区飯田橋3-10-10
　　　　　　　　　　ガーデンエアタワー18階
　　　　　　　　　　～～～～～～～～～～～～～～～～～～

ashimura, 2017-01-11:
Ah, got it.
You needed to substitute "/issue/" with "/issues/" in the URLs of the issues (s/issue/issues/) and have done so :)

peterMelco, 2017-08-16:
I think this can be closed.

closed 2017-08-16 by peterMelco