w3c/automotive #76: Subscribe should not return an integer

djensen47, 2016-01-06:
Returning an int allows for unsubscribe forgeries. Let's say my app has an ad network and a malicious ad gets onto the network, that ad could potentially brute force unsubscribe.

A better solution would be to return an object. Even better would be for that object to have the unsubscribe call itself.

tripzero, 2016-01-06:
Returning an object that had the unsubscribe method on it sounds reasonable.

Of course, you could also reduce the possibility of a malicious unsubscribe
by returning a significantly large random integer.  You could use web
crypto to generate the random number to take advantage of hardware
randomization.

tobie, 2016-02-04:
The security argument is moot the minute you're allowing third party code to execute within the same origin.

tguild, 2017-03-14:
Old issue for WebIDL

closed 2017-03-14 by tguild