w3c/automotive #77: [vehicle-spec] Are zones enough for all use cases

tobie, 2016-01-21:
A quick question as this is an issue I bumped into while working on the generic sensor spec.

Most cars have four proximity sensors on the rear bumper.

How would you target one of them given the current `Zone` proposal?

wonsuk73, 2016-02-05:
@tobie, Good point. Current ['Zone' Interface](http://rawgit.com/w3c/automotive/master/vehicle_data/vehicle_spec.html#zone-interface) couldn't cover the case you mentioned. Do you have any idea for this?

@aShinjiroUrata @acrofts84 @drkevg @tripzero @QingAn @paulboyes  Do you have any opinion or comment on this?

tobie, 2016-02-05:
@wonsuk73 accidental close?

aShinjiroUrata, 2016-02-05:
It seems no response yet, I write as my personal opinion. 
I think it is difficult to cover everything in a vehicle with `Zone` method since OEM can put as many sensors as they like, such as proximity sensor, camera, thermometer, etc.

I think `Zone` method is very good for vehicle specific items such as seats, tires, steering wheel, etc.
Proximity sensor, camera, thermometers are not vehicle specific things and they can be used in house or in many other situations. So API of those sensors can be standardized not as vehicle specific way but as more generic way.

1) So, one idea is, in Vehicle Data spec, just treat proximity sensors of very common numbers and positions which may be able to be handled by `Zone` method. 
If the vehicle has much more proximity sensors, they should be handled not from Vehicle Data but from Generic Sensor API or such other W3C API. 

2) Another idea is, to allow expanding `Zone` method so that OEM can use original position name such as 
```
var zone1 = new Zone();
var zone2 = new Zone();
//zone.value = ['front', 'left', 'bottom'];  //standard usage
//OEM's original name (implementation dependent)
zone1.value = ['prius-parksensor-rear-1'];
zone2.value = ['prius-parksensor-rear-2'];
vehicle.parkingSensors.get(zone1).then(function (sensor) {
    console.log("rear leftmost distanceToObject = " + sensor.distanceToObject);
})
```
Although I think this is not a standard.

Another idea will be something like divide 3x3x3 to 9x9x9 but this may no be convenient and this still has limitation...

wonsuk73, 2016-02-05:
@tobie, sorry to close. I didn't know that :( my mistake. Thanks for let me know.

tobie, 2016-02-05:
No problem!

yiyiamee, 2016-02-17:
Hello,
    
    Thanks for your email. I am on Chinese Spring Festival holiday from
    Jan 30 to Feb 16. I have limited access to emails during this
    period.
    
    I will be back to office on Feb 17.
    
    Thank you.
    
    BR, Yingying

tguild, 2017-03-14:
Zone approach abandoned as we moved to VSS JSON data tree

closed 2017-03-14 by tguild