w3c/automotive #78: Removing  "interface" suffix from the interface's name

wonsuk73, 2016-02-04:
Within the discussion for #72, @tobie and @peterMelco suggest to remove  "interface" suffix from the interface's name. 

Vehicle Information Access API spec has a below interface list for this issue.
- VehicleInterface Interface
- VehicleConfigurationInterface Interface
- VehicleSignalInterface Interface

After delete the suffix, those name might be replaced with as below.

- Vehicle Interface
- VehicleConfiguration Interface
- VehicleSignal Interface

It makes sense to me.

@acrofts84 @aShinjiroUrata @QingAn, @paulboyes , What do you think?

aShinjiroUrata, 2016-02-04:
I think one problem is, `Vehicle Interface` is already existing.
http://rawgit.com/w3c/automotive/master/vehicle_data/vehicle_spec.html#vehicle-interface

Current: 
- Vehicle Interface
- VehicleInterface Interface
- VehicleConfigurationInterface Interface
- VehicleSignalInterface Interface

So I suppose, new name should be like
- Vehicle Interface
- VehicleBase Interface (or other name?)
- VehicleConfiguration Interface
- VehicleSignal Interface

wonsuk73, 2016-02-05:
@aShinjiroUrata, Thanks for good point!
For the changing the name of VehicleInterface "VehicleBase" looks good to me!

@acrofts84 @QingAn @paulboyes What do you think?

peterMelco, 2016-02-05:
I agree, VehicleBase is good.

wonsuk73, 2016-11-08:
This issue is related with the vehicle information access API spec which is not developed in the group anymore.

closed 2016-11-08 by wonsuk73