w3c/automotive #80: Action Items to Refactor the Automotive API

djensen47, 2016-03-03:
As discussed in the meeting on Feb. 16

- [ ]  Discussion on how the service (websocket/rest-ish) approach works with strawman examples.
- [ ] Determine the architecture and diagram it. (Look at Web of Things).
- [ ] Convert WebIDL in the data spec to JSON or JSON-LD.
- [ ] Validate that a higher level API can be implemented using the newly proposed low-level API.
- [ ] Close #37 and #72 in favor of creating the new service-based API.

djensen47, 2016-03-04:
Started the discussion on #81 
Closed #37

wonsuk73, 2016-03-17:
@djensen47 I think Signal Naming Convention is also one of important item we need to make for websocket/rest-ish approach. What do you think?

closed 2017-03-14 by tguild