w3c/automotive #85: Format for Vehicle Information Access API Spec

pkinney, 2016-05-04:
Determine the format to use for specifying the Vehicle Information Access API:

* WebSocket protocol
* HTTP API routes

pkinney, 2016-05-17:
# Documentation Format/Style

At the bottom of this comment, I've listed out a few of the possible options for adding varying levels of strictness on the format of the documentation.  However, regardless of the format that we may choose to expose pieces or all of the documentation, I believe we need to generate a specification that W3C can publish and that follows those general patterns.  This means a "static" version.  We have the option to "export" some of the documentation in any number of more interactive formats, but will need the "core" static version first, I think.

To that end, I think we can decide on a content structure and some formatting details up front.  I propose the following TOC for the specification (stealing a bit form the current Vehicle Information Access Spec):

* Introduction/Puprose/Overview
  * Basic concepts
  * What this spec describes
  * What it means to be "compliant" (?)
  * etc.
* Terminology
* Security and privacy considerations
* Data Model (reference the VSS or resulting spec)
  * How the model is defined
  * How to reference the model
  * How to discover the model
* WebSocket
  * Connection Setup
  * Protocol Overview - who says what when
  * Message Types
    * subscribe
    * unsubscribe
    * get
    * set
    * ...
  * Use Cases and Sample/Reference code
* HTTP
  * Headers
  * Routes
    * GET /api/v1/...
    * PUT /api/v1/...
    * ...
  * Use Cases and Sample/Reference code

## WebSocket Messages

For each message, we need to define:

  * Description
  * Payload
  * Possible Responses
  * Possible errors

## HTTP API

For each route, we need to define:

  * HTTP Method
  * Path
  * Description
  * Parameters
  * Headers
  * Request Payload (for POST / PUT)
  * Responses
  * Possible Errors

paulboyes, 2016-05-17:
I think this is a great start.  I will gladly move to wiki to start filling it out.  From there we can go to respec

as2902b, 2016-05-18:
Still in the process of joining W3C Automotive.
However, I think it would be good to share some generic comments.
Considering that the formats used by VSS and OCF are based on YAML and its derivative RAML - I would prefer to stick to one of these formats, preferably RAML.
Developers love swagger and it is driving the Open API Initiative (https://openapis.org/).

rstreif, 2016-12-20:
Per Working Group Meeting: ticket stands; Vehicle Information API Specification yet to be created;

aShinjiroUrata, 2017-04-18:
VIAS document has been already created and work is progressing.

This discussion is old and not necessary any more.
I close this.

Since the "Documentation Format/Style" survey looks worthful, I copied it to wiki.
https://github.com/w3c/automotive/wiki/Documentation-Format-Style-Survey-for-VIAS

closed 2017-04-18 by aShinjiroUrata