w3c/automotive #87: Charter Issues - Deliverables Defintions

acrofts84, 2016-07-28:
There are a couple of contentious issues in the current charter:

- Where would we like to place the dependence on the VSS? On both specifications, one or elsewhere in the specification?
- Is a VSS snapshot an other deliverable?
- In the client spec definition we claim it can be implemented without adhering to the server specification. Is this appropriate?
- Do we want to define the Vehicle Signal Client  Spec as the Vehicle Signal Client API?

My opinion is that: 
- the dependencies should be on both specs
- the snapshot is not a w3c deliverable
- the client spec should not be suggested for usage without the server spec
- Client spec should be the Vehicle Signal Client API

paulboyes, 2016-07-29:
I think....

- VSS dependencies should be in both specs.
- VSS is a dependency not a deliverable
- Client spec carries less weight if totally dependent on service spec.  Abstraction layers are good.  However I understand the argument.
- Changing the name of the client spec to something different from the old spec name makes sense to me.  Client signal spec is good.

rstreif, 2016-12-20:
Per Working Group Meeting: addressed at this point with the new charter. Close.

closed 2016-12-20 by rstreif