w3c/automotive #88: Create demo for Genivi AMM meeting Burlingame

peterMelco, 2016-08-17:
We need to decide if we should do this and what the content of the demo should be. The goal and purpose needs to be defined.

I have started a on a description on:
https://www.w3.org/auto/wg/wiki/Main_Page#Oct_20_-_21_.2C_2016_.2CGenivi_AMM.2C_Burlingame

paulboyes, 2016-08-23:
This was discussed during the GENIVI/W3C Liaison Meeting this morning.  I think Peter has done a great job of describing the bare basics.  Is anyone willing to and able to commit to working on this?

During today's meeting we reserved a table for the GENIVI AMM Showcase.  Whether we do the demo or not, we should be prepared to man the table and let people know what the group is up to.  

That said, demo(s) would be great.

newskysecurity, 2016-08-24:
I spent a few hours and setup a sample server:

https://auto.newskysecurity.com/chat.xhtml

The server is currently running a sample WSS-based web chat, demonstrating WSS protocol communication. For the four requirements on the server side:
 - PC simulation 
 - WSS protocol
 - subscribe, set
 - Token based authorization and error handling
This server shows the first 2 are feasible.  I can work more on the latter two requirements.

newskysecurity, 2016-08-26:
spent another few hours and created a demo page showing subscribing to a vehicle signal using WSS and get data stream back. 

https://auto.newskysecurity.com/speed.xhtml

Usage:

1. Click 'connect'
2. Click 'Send message'
3. Watch subscribed vehicle speed data (simulated) coming back from WSS every 3 seconds

Supported features:

1. Basic authentication via token, hard-coded auth string 'AuthMagic', use a different value and server will return 403
2. Hard-coded JSON format. 
3. Subscription to vehicle signal. Only supports root.engine.speed. Once subscribed, will keep receiving simulated speed data every 3 seconds
4. Unsubscription to vehicle signal. Send  {'auth':'AuthMagic', 'unsubscribe':'root.engine.speed'}  and the speed data stream will stop

Known bugs / TODOs:
1. Vehicle speed is not singleton, open two browser tabs and subscribe to the same server, you won't see the same speed value.
2. TODO: use json parser to be more flexible on input / output.
3. TODO: more flexible auth levels?
4. TODO: add a better UI

Pls let me know if this is the right direction...

peterMelco, 2016-11-15:
Well, this is done

closed 2016-11-15 by peterMelco