w3c/automotive #104: trim abstract

dret, 2016-10-21:
it seems that the current abstract is more an introduction than an abstract. it might help the spec's visibility to have a shorter, one- or two-paragraph abstract.

wonsuk73, 2016-10-22:
Thanks for comment! @dret, I agreed. One or two paragraph abstract is typical shape of W3C spec.

dret, 2016-10-23:
as a side note: http://www.w3.org/TR/2016/WD-vehicle-information-api-20160119 has a similarly long abstract. not quite as long but maybe also worth trimming.

dret, 2016-10-25:
On 2016-10-22 11:42, Wonsuk Lee wrote:
> Thanks for comment! @dret <https://github.com/dret>, I agreed. One or
> two paragraph abstract is typical shape of W3C spec.

i'd be more than happy to help with reviewing/tweaking those paragraphs. 
i first have to read through the whole thing, though. thanks for the 
quick feedback!

wonsuk73, 2016-11-08:
@dret I had updated abstract part! Thanks for the comment!

closed 2016-11-08 by wonsuk73