w3c/automotive #107: discrepancy between title and short title

dret, 2016-10-23:
currently the title is "Vehicle Signal Server Specification" and the short title is "vehicle-information-service" (server vs. service). to me, service makes much more sense (no consumer is interested in the appliance providing a service), but whatever it is, it should be consistent.

rstreif, 2016-12-20:
Per Working Group Meeting: rename title to "Vehicle Information Service Specification"

dret, 2016-12-20:
looks good to me!

rstreif, 2016-12-20:
In addition per Working Group Meeting: name the companion document specifying the API "Vehicle Information API Specification"

drkevg, 2017-01-04:
Updated name of server spec to be 'Vehicle Information Service Specification' and for references to client spec to 'Vehicle Information API Specification

closed 2017-01-04 by drkevg