w3c/automotive #121: Wrong document listed first when Google 'Vehicle Information Service Specification'

drkevg, 2017-01-06:
Spotted that if we google 'Vehicle Information Service Specification' the first match that google provides is to our very old rough working document - which is very misleading:
https://www.w3.org/auto/wg/wiki/Vehicle_Information_Service_Specification

The current version of the specification, with the new name is at:
https://w3c.github.io/automotive/vehicle_data/vehicle_information_service.html

Is it possible to re-issue the FPWD with a new publication date so that the title is corrected as it currently has the old title (please see below)
https://www.w3.org/TR/2016/WD-vehicle-information-service-20161020/

rstreif, 2017-01-06:
Good catch @drkevg.

For the Google search, I am not sure how we can influence the search engine. Should we retire, meaning remove, the old specification by moving it to a different URL and have the URL forwarded to the new spec?

As far as reissuing the FPWD is concerned, our process specialists should be able to tell us if that's an option. However, we have the CR planned for April. It is expected that we have modifications to the specification as a result of reviews done by others and us in response to the FPWD. Hence, wouldn't it be reasonable to make the changes, publish them via Github as we go and then issue the CR in April?

drkevg, 2017-01-09:
Paul has updated the following document with explanatory note and links to the FPWD and the WD
https://www.w3.org/auto/wg/wiki/Vehicle_Information_Service_Specification
Issue can be closed.

closed 2017-01-09 by drkevg