w3c/automotive #123: Implementation Report Plan

Hirabayashi-KDDI, 2017-01-10:
I have placed a draft (pptx version 01) of Implementation Report Plan (IRP)　as a new issue  on the GitHub.　In a few days, web style document will be available.
If any, please put comments on the issue.

Hirabayashi-KDDI, 2017-05-09:
I generated VISS & VIAS Implementation Report Plan in web format, based on new drafts of VISS and VIAS .
[VISS&VIAS Implementation Report Plan.txt](https://github.com/w3c/automotive/files/988250/VISS.VIAS.Implementation.Report.Plan.txt)
[VISS&VIAS testing.zip](https://github.com/w3c/automotive/files/988258/VISS.VIAS.testing.zip)

peterMelco, 2017-08-16:
I think this one can be closed . We will have a wiki page where we link to implementation reports. The content and format of the actual reports are not stated. It is, howeve,r good if the reports reflect upon the spec and pitfalls found during the implementation.

closed 2017-08-16 by peterMelco