w3c/automotive #127: Make introduction non normative

drkevg, 2017-01-11:
Comparing with old one, I feel that the introduction part of the current spec is a little too specific.
In my understanding, the introduction part is usually non-normative (So key words such as MAY/MUST/SHALL does not appear.)
Particularly, I would like to suggest that paragraphs refereeing websocket connection could be moved to the security section (initially raised by Junichi via email 11Jan17)

rstreif, 2017-01-11:
I agree with @drkevg. The Introduction section contains too much detail on architecture and implementation. I feel that these details should actually go into the Architecture section. In addition, the Architecture section again has an Introduction subsection which should probably be renamed to something like "Overview". @drkevg do you want to take a stab at the improvements you suggest?

paulboyes, 2017-01-12:
Giving credit where credit is due, this comment originated via email from Junichi Hashimoto.  Kevin graciously created the issue and hopes to resolve in the next couple of days.  "Thanks very much for the helpful feedback, I've raised issue #127 to capture this. Adam and I would hope to resolve in the next couple of working days."

sou3ilow, 2017-01-12:
All, thank you for adressing this issue.

rstreif, 2017-03-14:
AR: Editors please verify if this request has been fulfilled. If not, please address accordingly.

closed 2017-05-09 by acrofts84