w3c/automotive #130: Review of [Vehicle Client Specification](https://github.com/w3c/automotive/blob/gh-pages/vehicle_signal_client/vehicle_signal_client_spec.md)

rstreif, 2017-01-12:
My initial review comments:

* Client Goals (line 7): "VISS server" should be "VIS sever" as the last S stands for "Specification".
* W3CClient Object (line 27ff): The name W3CClient is rather generic and nondescript. I suggest VISClient.
* filter (line 84): "The structure of this option may be server-specific and [is out of scope of this specification?]": if it is server-specific and not defined, neither by this specification or VISS, then we most likely will get a problem with interoperability. I suggest that we define the filter format in VISS and reference it from this spec.
* Example code (line 114): ${_client.protocl} should be ${_client.protocol}
* Example code (line 218): The onConnect function should be openWindow as defined above, not openAllWindows.

pkinney, 2017-02-02:
@rstreif: Thank you very much for your review!  PR #134 submitted to address all of the above except the expansion on filter.

1. Done
2. Agree completely; was waiting on a better name and I think `VISClient` is much better.
3. Agree.  Will reference once that is in the VISS spec.
4. Done
5. Done

rstreif, 2017-02-02:
Thank you, @pkinney. I am good for this round of review.

aShinjiroUrata, 2017-04-03:
Fix for this comment have done in below PRs.
- PR #165 : Issue #163, rename from 'W3CClient' to 'VISClient'.
- PR #166 : Changed word 'VISS server' to 'VIS server'. And minor fix.

This was originally, @pkinney trying to fix via PR #134 , however I fixed
 with different PRs. I closed PR #134 without merging.

closed 2017-04-03 by aShinjiroUrata