w3c/automotive #132: What fomat of json should be returned by getVSS()?

aShinjiroUrata, 2017-01-17:
One possible format option should be the one defined in VSS spec which includes 
 a lot of information such as type, description, min, max, unit, etc. 
https://github.com/GENIVI/vehicle_signal_specification/blob/develop/vss_rel_1.json

Another option should be something more light weight one which includes only necessary information.

I guess both may be possible but I suppose better to define in the spec.

rstreif, 2017-01-17:
My suggestion would be to define a flag as part of the API that provides selection of the format. In its simplest form that could be a boolean flag for a short form that just provides the signal paths and the ID and a long form that provides all information.

aShinjiroUrata, 2017-01-18:
>My suggestion would be to define a flag as part of the API that provides selection of the format.

Thanks for the comment!
This looks practical solution.

Reason why I raised this issue is, I first though returning simple JSON like below.
But in this case, it may be difficult to tell a node is a 'leaf' or a 'branch with no children'.
So I thought 'type' information may be necessary.

```
{Signal:
  {DriveTrain:
    {InternalCombustionEngine:
      {RPM:
        {type:UInt16
         min:0,
         max:20000,
         id:54,
         unit:'rpm'}
      }
    }
  }
 Attribute:{},
}
```

On the other hand, using VSS json is like below
 and this include necessary information.
But I felt this might be too much such as, 
 I don't want 'children' node and other not necessary information for client.
(Of course, not a big problem. It will be OK by just ignoring 'children' node
 and other attributes. 
 Data size will also be not a big trouble since getVSS() may not
 be used too frequently.)
```
{Signal:
  {type:branch,
   description:'xxx'
   children:
    {DriveTrain:
      {type:'branch',
       description:'aaa',
       children:
        {InternalCombustionEngine:
          {type:'branch',
           description:'aaa'
           children:
            {RPM:
              {type:UInt16
               min:0,
               max:20000,
               id:54,
               unit:'rpm'}
            },
          }
        },
      }
    }
  }
}
```

And I wanted to ask someone has any idea or preference on this.

rstreif, 2017-01-18:
@aShinjiroUrata In your first example, which is perfectly fine with me, you can tell if a node is a leaf if `type != "branch"`. However, I would suggest that we stick to the second format as it is the format that the [VSS JSON converter](https://github.com/GENIVI/vehicle_signal_specificationl) outputs. Reason is not a mere format preference but simply only needing one parser implementation.

Either way, it might be more useful to have a less verbose format that only returns the ID rather than type, min, max, unit, description. All subsequent queries could then be made simply by using the ID which should be much quicker. That is essentially how [VSI](https://github.com/GENIVI/vehicle_signal_interface) works. You can query signals by name but also by ID. There are two b-trees, one organized by name and one by ID. However, the ID comparison will always be faster as it's just integer and not a string as the name.

aShinjiroUrata, 2017-01-19:
@rstreif, thanks for the comment again.

> However, I would suggest that we stick to the second format as it is the format that the VSS JSON converter outputs. Reason is not a mere format preference but simply only needing one parser implementation.
 
I agree that better to use only one JSON format than using two.
Everyone is OK, I'm OK.

> You can query signals by name but also by ID. There are two b-trees, one organized by name and one by ID. However, the ID comparison will always be faster as it's just integer and not a string as the name.

I understand. Using path for query has disadvantage on data length as well.

acrofts84, 2017-01-24:
@aShinjiroUrata @rstreif, thank you both for clarifying this behaviour. I have updated the spec to reflect this in pull request #133

acrofts84, 2017-01-25:
As discussed, I've only included the verbose version in the pull request. However, the benefit of the verbose version is probably only during development, as you will not validate against 'description' programmatically and, as you've said, 'type' can be implicit. The description may also add a lot of overhead. 

The second parser on the server side is probably worth the benefit of having the more succinct VSS version. 

The key question is how much data to omit, and what is going to be useful at runtime. I think the 'min', 'max', 'type', 'id' and 'units' will be useful to have, as per @aShinjiroUrata's example. The <a href="https://github.com/w3c/automotive/blob/gh-pages/vehicle_signal_client/vehicle_signal_client_spec.md">client spec</a> has, so far, specified pathsByCSS(cssSelector), pathExistsByCSS(cssSelector), canGet(path), canSet(path), at([path]). These require an indication of permissions (read, write, visible, invisible) in the VSS structure, which I cannot currently see in the VSS spec. @rstreif, has this been provisioned in the VSS structure?

rstreif, 2017-01-25:
@acrofts84 The VSS currently does not have a flag to indicate the access to a particular signal. Adding that does make sense on a general level. For instance all attributes are read-only as are some signals such as Drivetrain.Engine.RPM. Other signals are read and write for example Body.ExteriorLights.IsHighBeamOn (although the name may suggest status only).

Actual permissions as set by the access control token may modify that. The question is how this modification should work. Here is a stab at it:
| VSS Access Flag | Access Control Token | Result                    |
|-----------------------|-----------------------------|--------------------------|
| not set                 | any                              | no access                |
| read only             | not set                         | no access                |
| read only             | read                            | read only                  |
| read only             | write                            | no access                 |
| read only             | read/write                    | read only                 |
| write only             | not set                         | no access                |
| write  only            | read                            | no access                |
| write only             | write                            | write only                 |
| write only             | read/write                    |write only                 |
| read/write            | not set                         | no access                |
| read/write            | read                            | read only                  |
| read/write            | write                            | write/only                 |
| read/write            | read/write                    | read/write                 |

The next question is if a visibility flag is necessary. I do not think so as access control also controls visibility. If there is no access then the signal should also not be visible.

acrofts84, 2017-02-03:
Thanks @rstreif. The tabular approach you have suggested looks good to me. 
The visibility flag is an interesting one. The benefit of have a visibility flag is that it will allow the app to see what is available, given elevated privileges, at which point it can request a token from the auth server or ask the user to login, and therefore elevate user privilege. If it is not visible, then the app is not going to be able to access this signal, so there is no point in trying these further steps.

rstreif, 2017-02-03:
@acrofts84 My thinking was to derive the visibility from the access, meaning if there is "no access" in the table above it would not be visible. I agree that an app could use the knowledge about the availability of a signal to request a certificate that grants access from a server. However, that could pose additional security challenges. I followed a more conservative path according to the German proverb "Was ich nicht weiss, macht mich nicht heiss." or its English equivalent "What the eye does not see, the heart does not grieve over." :) Nevertheless, I am not opposed to adding a visibility flag.

tguild, 2017-02-08:
http://www.w3.org/2017/02/08-auto-minutes.html

tguild, 2017-02-21:
read/write flags in VSS pending so reopening
https://www.w3.org/2017/02/21-auto-minutes.html

rstreif, 2017-03-14:
AR: Amend "VSS Request" section to clarify that getVSS returns all signals of the tree or a subtree regardless of any access rights granted by authorization tokens. It is the duty of the client to know what signals it can access. If a signal is requested to which the client has no access then the server will reject the request as detailed by the Get, Set, and Subscribe sections of the specification.

acrofts84, 2017-05-09:
Thanks Rudi. Here is my proposed wording:

The server shall return VSS metadata for all branches and nodes in the VSS tree, with the exception of the "Private" branch, regardless of whether the client can access them under the current access controls. If a signal is requested to which the client has no access then the server will reject the request as detailed by the Get, Set, and Subscribe sections of this specification. The server shall not return the metadata for the branches and nodes within the "Private" branch unless the client has appropriate access controls.

acrofts84, 2017-05-25:
@rstreif Does my proposed text align with your understanding of how getVSS should work?

drkevg, 2017-05-31:
Agree with Rudi that: 

_"getVSS returns all signals of the tree or a subtree regardless of any access rights granted by authorization tokens. It is the duty of the client to know what signals it can access. If a signal is requested to which the client has no access then the server will reject the request as detailed by the Get, Set, and Subscribe sections of the specification."_

And with Adam that:

_"The server shall return VSS metadata for all branches and nodes in the VSS tree, with the exception of the "Private" branch, regardless of whether the client can access them under the current access controls. If a signal is requested to which the client has no access then the server will reject the request as detailed by the Get, Set, and Subscribe sections of this specification. The server shall not return the metadata for the branches and nodes within the "Private" branch unless the client has appropriate access permissions."

Reason is mainly that this is simple and the VISS server is effectively operating in the same way as an OS or WebServer granting access to resources. If a client makes a request that it is not authorized to perform it gets an error e.g. 403 Forbidden. It would be very complex to list all of the actions that different users or roles could perform and also this information could be of benefit to an attacker.

acrofts84, 2017-06-20:
Working Group Call - agreed new wording

closed 2017-07-18 by acrofts84