w3c/automotive #136: Consider adding filtering to Set e.g. to set a subset of doors so that isLocked=true

drkevg, 2017-02-06:


rstreif, 2017-03-14:
It's probably "wildcarding" rather than "filtering". This example from the specification:

client -> {
		"action": "set",
		"path": "Signal.Cabin.Door",
		"value":{ [ {"Row1.Right.IsLocked" : true },
		            {"Row1.Left.IsLocked" : true },
			    {"Row2.Right.IsLocked" : true },
			    {"Row2.Left.IsLocked" : true } ] },
                "requestId": "5689"
		}

means that a developer would have to know how many rows and doors there are and explicitly ask for every single one to be locked. A better way would be allowing wildcards in the path:

client -> {
		"action": "set",
		"path": "Signal.Cabin.Door.*.*",
		"value":{ [ {"IsLocked" : true } ] },
                "requestId": "5689"
		}

AR: @drkevg please clarify if this is what you meant.

drkevg, 2017-05-10:
Suggestion withdrawn. Requires wider consideration so will discuss and may re-raise if thought to be an improvement.

closed 2017-05-10 by drkevg