w3c/automotive #141: Subscribe repsonse inconsistent

peterMelco, 2017-03-13:
**Section: 8. Message Structure 
Cat: Subscribe**

When subscribing to signals the action attribute is not part of the return JSON message, this is inconsistent . The inconsistency makes testing ugly.

Suggestion: add "action", "subscribing" to respond message.

acrofts84, 2017-03-13:
Hi Peter. Thanks for the comments. I agree that it is ugly. Urata-san mentioned this in #138, where I had assumed that the subscriptionNotification would return the action: "subscribe".

I agree that this is ugly and we should include something here. Maybe action = "subscription" or "subscriptionNotification"

rstreif, 2017-03-14:
AR: Resolve the issue as suggested by this issue and issue #138.

peterMelco, 2017-03-14:
Sounds good.

acrofts84, 2017-03-28:
Action "subscription" added and merged

closed 2017-03-28 by acrofts84