w3c/automotive #142: When using a wild card ( * ) as part of a path it is unclear how to specify each corresponding value.

peterMelco, 2017-03-13:
**Section: 8 Message Structure
Cat: Set**

The example below is from the spec: 
```
"action": "set",
"path": "Signal.Cabin.Door.*.IsLocked",
"value": { [ {"Row1.Right.IsLocked" : true }, {"Row1.Left.IsLocked" : true }, {"Row2.Right.IsLocked" : true }, {"Row2.Left.IsLocked" : true } ] },
```
It appears that all values corresponding to the * are listed. What if only three out of four possible values are listed - is that an error case? In that case, why have a wild card, when all values still must be listed? Seems redundant.

rstreif, 2017-03-14:
Agreed, this is somewhat confusing. In this example the path actually should just be "Signal.Cabin.Door" and should be extended by the specific values:

"action": "set",
"path": "Signal.Cabin.Door",
"value": { [ {"Row1.Right.IsLocked" : true }, {"Row1.Left.IsLocked" : true }, {"Row2.Right.IsLocked" : true }, {"Row2.Left.IsLocked" : true } ] },

AR: Correct the example in the specification.

peterMelco, 2017-03-14:
So, to set one leaf value we would use the following notation ?

```
"action":"set",
"path":"Signal.Cabin.Door.Row1.Right.IsLocked",
"value":true
```

rstreif, 2017-03-14:
Yes, although I am not quite sure about the JSON. Even though it's a single element it might need to be like this to make it consistent for the parser:
```
"action":"set",
"path":"Signal.Cabin.Door.Row1.Right.IsLocked",
"value": {[{true}]}
```

acrofts84, 2017-05-09:
Hi All, 
I think we could take this in a different direction and provide a couple of possibilities.

<pre>
// Set all IsLocked statuses within Signal.Cabin.Door to true
{
      "action": "set",
      "path": "Signal.Cabin.Door.*.IsLocked",
      "value": true,
      "requestId": "5689"
}
</pre>

<pre>
// Set the IsLocked status of specific doors
{
      "action": "set",
      "path": "Signal.Cabin.Door",
      "value": [ {"Row1.Right.IsLocked": true },
                 {"Row1.Left.IsLocked": true },
                 {"Row2.Right.IsLocked": true },
                 {"Row2.Left.IsLocked": true } ],
      "requestId": "5689"
}
</pre>

I agree we should add an example with an error code, were the doors cannot be set to locked (e.g. since they are open). Maybe a 409 Conflict error?

acrofts84, 2017-05-10:
- We need to queue set requests on the server until the response is sent
- 4XX implies client error, is this more a 5XX? 502 Bad gateway?
- Do we prevent setting multiple values until we have an appropriate solution to the case where one door is unable to be locked, where the other doors are locked. Should we only allow one signal to be set per request?

wzr1337, 2017-05-11:
Will there be a "rollback scenario" be defined, in case the query is only partially successful?

acrofts84, 2017-05-31:
Agree to set one thing at a time - i.e. no wildcards, single leaf nodes only:
- Simplest solution
- The client has full visibility of errors
- The one after another "knight rider" effect is unlikely to occur given the latencies involved
- Errors can be handled individually, but still grouped by a client's abstraction layer
- Individual handling of errors means the client can decide to abandon, e.g. unlocking all doors if one cannot be locked. i.e. client can decide how to handle rollback scenarios

Agree that we need to queue set requests on the server

Agree error messages suggestion:
- 502 Bad Gateway is best choice as matches the error case 'The server was acting as a gateway or proxy and received an invalid response from the upstream server'

@drkevg (also agree with comments above)

acrofts84, 2017-06-20:
WG Meeting:
It is most efficient to allow the client to request multiple values to be set using a wildcard. Most of the time this will follow the happy path and will provide no issue. When there is an error with 1 or more signals being set, we will return an error code. It is the client's responsibility to query the current state of the signals following this error code.
The set section of the spec  requires an update to describe this behaviour

peterMelco, 2017-08-16:
So, the consensus is that we use wildcards to set muiltiple values, and values that are not set arereflected in an error response ?

peterMelco, 2017-09-05:
I will close this one.

closed 2017-09-05 by peterMelco