w3c/automotive #143: For filter "range", the wording "above" and "below" is misleading.

peterMelco, 2017-03-13:
**Section: 9 Server side filtering
Cat: Filter**

The example below from spec:
//client receives data when the value is between 100 and 200 (inclusive)
` { "action": "subscribe", "path": "<any_path>",          "filters": { "range": { "above": 100, "below": 200 } },          "requestId": "<some_unique_value>" }`
To specify "above" 100 sounds like the first value to send is 101. And "below" 200 sounds like 199 is ok but not 200. Proposal: use "min" and "max" instead.

acrofts84, 2017-03-13:
Hi Peter. The reason behind this wording was to allow the two-tailed case, where data is returned for small or large values. I.e. I want to know when engine oil temperature is below 5 degrees and above 100 degrees, but I don't care in the middle. Min and max would be confusing for this use case, therefore we thought Above and Below explained it better. We have stated that this should be inclusive, but I agree that the words do not imply inclusive

rstreif, 2017-03-14:
No action. The words "above" and "below" are specifically used with the word "inclusive" to indicate an range. the words "min" and "max" are used to indicate single thresholds.

closed 2017-03-14 by rstreif