w3c/automotive #144: Filter behaviour unclear

peterMelco, 2017-03-13:
Section: Server Side Filtering
Cat: Filter

What is the purpose of the filter behavior specified below (provide example use case)?
"When the range filter is used a final message is sent when the value returned is outside of the specified range. For example, if the range states { "below": 100 }, a final value may be received at 101 to indicate that the value is now out of range."

We need to provide a use case or a better explanation on this behaviour.

acrofts84, 2017-03-13:
Hi Peter. The use case for this is when the value of a signal is in an unreasonable range. If I want to receive a notification when my oil temperature is in a dangerous range, i.e. above 100 degrees, I will also want to know when the value leaves this dangerous range so we need to receive a value when the signals leaves the range.

rstreif, 2017-03-14:
The specified behavior is somewhat confusing and would cause more issues if the value jumps. The client is expected to be smart about when not receiving any updates over a period of time which means for the value to be out of range.

AR: Remove the paragraph "When the range filter is used a final message is sent when the value returned is outside of the specified range. For example, if the range states { "below": 100 }, a final value may be received at 101 to indicate that the value is now out of range." from the specification.

drkevg, 2017-05-09:
Agree we can remove this. The original intention was to ensure that the client was informed if the value moves out of the range being monitored. This was to enable the client to detect  the difference between 'out of monitored range' scenario and the server simply being slow in responding. However, in many cases the client won't care and in cases where it does, the client could create a second subscription that only fires when the value moves out of the range. Hence agree we can remove this.

closed 2017-05-09 by acrofts84