w3c/automotive #146: D) 4. W3CClient() > Functions: authenticate() method name and other issues.

aShinjiroUrata, 2017-03-21:
Seprarated from #137 

==================== 
@aShinjiroUrata wrote at 2017 Feb22

D) 4. W3CClient() > Functions: authenticate()

(1) 
In VISS, word 'authorize' is used as method name.
In VIAS, method name 'authenticate' is used.
May be better to align to ’authorize’ or 'authenticate'?

According to below URL, in authentication, usuallly userID, password are used.
http://stackoverflow.com/questions/6556522/authentication-versus-authorization
So in this case, I guess, word authorize fits better?

(2)
In VIAS spec there are two `token` descriptions. Those are
- `authenticationTokens` at, 4. W3CClient Object > Client Options
and
- `token` at,  4. W3CClient Object > Functions > authenticate.

These tokes are same type of things?
Or one is authenticationToken and the other is authorizationToken or something?

(3)
According to current VIAS, `authorize` method returns `value`.
What object will be contained in `value`?

====================
 @pkinney wrote at 2017 Mar8

D) "Authorize" is the appropriate term as as I am "authorizing" this connection for certain access that it didn't have before. The Authentication of the token I am using is up to the client's internal implementation since at this point I have already "Authenticated" myself with the token-issuing authority in order to get the token. If there is confusion, perhaps we can change the function name to applyToken or some such thing.

peterMelco, 2017-03-22:
applyToken would the basically mean use this token to authorize an action ?

aShinjiroUrata, 2017-03-22:
Hi Peter-san, thanks for the comment.

>applyToken would the basically mean use this token to authorize an action ?

This is not answer to your question but the issue I raised originally was just about the difference of
 method name.
- in VISS: "Authorize" method
- in VIAS: "Authenticate" method
although these two are doing same thing.

So, I just want to close this issue by changing VIAS' method name from "authenticate" to "authorize" for
 the purpose of using same wording between VISS and VIAS.
This is the solution for 1).

2) and 3) are not clear description for me and I just wanted to confirm with Powell-san but I suppose
 we can not get the answer anymore, and so I just take back 2) and 3) questions.

The Powell-san's idea of using name "applyToken" create difference with VISS as well.
So, if we use "applyToken" in VIAS, we should consider to use "applyToken" in VISS as well.
But I think this in not necessary.

rstreif, 2017-03-22:
@aShinjiroUrata , @peterMelco ,

"authenticate" and "authorize" are two different things although in the context of VIS we do them both with one token.

"Authentication" is the process of proving that an entity actually claims who it is.
"Authorization" is the process of giving access to an entity.

For example, a pin code to a door lock with keypad authorizes any person knowing the pin to access the room behind the door. A biometric door lock with fingerprint reader authenticates a person through the fingerprint and also authorizes the person to unlock the door at the same time. A door lock with fingerprint reader and keypad, requires a person to authenticate with a fingerprint and to authorize unlocking door by entering the correct pin.

I don't necessarily agree with @pkinney wrote in D) above. Even though the entity could have authenticated itself with the issuing token-issuing authority, the VIS server (or the component handling the security on its behalf) needs to be able to authenticate and authorize the client. Authenticate in this case means that the server can verify that the client presenting the token is indeed the entity the token-issuing authority has issued the token to. According to the example above, the keypad door lock cannot identify that the person who is entering the code is the rightful owner of the code. The keypad with biometric scanner and keypad can.

So maybe a generic name for the API and function is a better choice to avoid confusion.

peterMelco, 2017-03-23:
@rstreif @aShinjiroUrata  Ok, Rudi the authentication of the client would then have to be handled by the token - some code that would identify the client server side ? 

In that case I guess that applyToken is generic ?

rstreif, 2017-03-23:
@peterMelco 

> So, the token signature should? be used - JWT as an example - to do the authentication, and the 
> authorization is done through token claims .
Essentially yes, the token authority should sign the token claims and an identifier of the client such as the client's public key. The token authority places the signature inside the token. The client signs the token with its private key. The receiving server then first uses the token authority's public key (which it has stored or can securely obtain) to validate the token claims and client they belong to. It then extracts the client's public key from the token and validates the client's signature of the token.

wonsuk73, 2017-03-27:
I think in the OAuth world the term of access token is typically used. Refer [the link](https://www.google.co.kr/search?q=authized+token&oq=authized+token&aqs=chrome..69i57j0l5.5382j0j7&sourceid=chrome&ie=UTF-8#q=access+token&*).

rstreif, 2017-03-27:
We should call the token "access token" as it contains an identifier for the entity and a list of privileges the entity has. What do we want to call the function that takes the "access token"? The proposal was applyToken(). Other suggestions: presentToken(), provideToken().

aShinjiroUrata, 2017-04-04:
Hi,

In tMarch27 VIAS call, there was discussion about this issue.
To summarize the opinions, it would be as below.
> - #146: Method name issue. 'authorize', 'authenticate' or 'applyToken', etc. disscussion on going.
>   - 'Authorize': Adam, Peter, Urata vote for this.
>   - 'access token'(?): proposal from Wonsuk by referring to OAuth word. Not strong preference.
>   - 'applyToken': Rudi-san

In the previous message, Rudi-san wrote as 
> Even though the entity could have authenticated itself
>  with the issuing token-issuing authority,
>  the VIS server (or the component handling the security on its behalf)
>  needs to be able to authenticate and authorize the client.

I think, Rudi-san's idea is, the "to-be-named" method (autorize/authenticate/applyToken)
 has role of "authenticate" and "authorize" both.
So, neither of "authenticate", "authorize" fits as method name.

And I think, Adam-san, Peter-san, (and Urata) idea is "to-be-named" method
 has role of "authorize" only since "authenticate" is done when
 token is issued by token-issueing authority.
So the name 'authorize' is appropriate.

I think this is the point of difference and makes different opinions.
What do you think?

By the way, this discussion started as VIAS issue but I think
 this is not only about VIAS but affects to VISS as well.
(The 'Authorize' method name in VISS might be changed)

rstreif, 2017-04-04:
Thank you, Urata-san.

I have inlined my comments below.

> Hi,
>
> In tMarch27 VIAS call, there was discussion about this issue.
> To summarize the opinions, it would be as below.
>
>
>    - #146 <https://github.com/w3c/automotive/issues/146>: Method name
>    issue. [...]
Urata-san, you are correct about what I am looking for. Authentication and
authorization are different things:

   - Authentication - validate that the entity actually is who it is saying
   it is
   - Authorization - validate the privileges of the entity

For the VIS use case authentication and authorization must be self-carried,
meaning the token presented by the entity

   - must contain verifiable information about who the entity is and that
   it is actually the correct entity who has presented the token, and
   - must contain the privileges

In that sense I do prefer a name

   - that does not imply either one of them but is neutral e.g. applyToken,
   logon, etc. then the documentation can explain that it is use for
   authentication and authorization with the same token, or
   - implies both such authenticateAndAuthorize

It might be a little bit of hairsplitting but I personally prefer exact
naming conventions, however, we can stick with 'authorize'. That's ok. I
just wanted to point out that it has to be more than just authorization.

> And I think, Adam-san, Peter-san, (and Urata) idea is "to-be-named" method
> has role of "authorize" only since "authenticate" is done when
> token is issued by token-issueing authority.
> So the name 'authorize' is appropriate.
Yes, the entity also has to authenticate itself with the token-issuing
authority (TIA) which then provides the token if it successfully can
identify the entity. However, it is not sufficient if the TIA simply
provides the token to the entity, it must ensure that the token cannot be
used by another entity, even if that other entity is also an entity known
and authenticated by the TIA. That means the token has to carry the
authentication validated by the TIA.

> I think this is the point of difference and makes different opinions.
> What do you think?
>
> By the way, this discussion started as VIAS issue but I think
> this is not only about VIAS but affects to VISS as well.
> (The 'Authorize' method name in VISS might be changed)
Correct.

:rjs

> <https://github.com/w3c/automotive/issues/146#issuecomment-291390913>, or mute
> the thread

-- 
*Rudolf J Streif*
System Architect - Open Source Initiative
Open Source Technology Centre

*M:* +1.619.631.5383
*Email:*  rstreif@jaguarlandrover.com

UK: G/26/2 G02 Building 523, Engineering Centre, Gaydon, Warwick, CV35 ORR
US: 1419 NW 14th Ave, Portland, OR 97209
jaguar.com | landrover.com
-------------------
Business Details:
Jaguar Land Rover Limited
Registered Office: Abbey Road, Whitley, Coventry CV3 4LF
Registered in England No: 1672070

This e-mail and any attachments contain confidential information for a
specific individual and purpose.  The information is private and privileged
and intended solely for the use of the individual to whom it is addressed.
If you are not the intended recipient, please e-mail us immediately.  We
apologise for any inconvenience caused but you are hereby notified that any
disclosure, copying or distribution or the taking of any action in reliance
on the information contained herein is strictly prohibited.

This e-mail does not constitute an order for goods or services unless
accompanied by an official purchase order.

rstreif, 2017-04-05:
As per discussion during the working group teleconference on 4/4/2017, let's name the method `authorize` in VISS and VIAS.

INRIX-paul-boyes, 2017-04-05:
I concur with Rudi

aShinjiroUrata, 2017-04-06:
According to above conclusion, created PR.

drkevg, 2017-04-06:
I also concur with Rudi that we name the method 'authorize' in both VISS and VIAS.

Believe Powell is also right when he says "Authorize" is the appropriate term as as I am "authorizing" this connection for certain access that it didn't have before. " The mechanism is being used to pass tokens that identify particular entites e.g. the user or device  

This is a confusing area though - for a couple of reasons. 

Firstly, by the time we call the 'authorize' method, the entities (user, vehicle etc) whose tokens are passed via the authorize method must have been identified (authenticated) by some trusted security authority (the one providing the tokens). Then after the tokens are passed via the 'authorize' method,  the server is responsible for checking with the Security Authority that issued the token, that the token is valid.

From the server's perspective, before the 'authorize' method is called, the server is free to implement whatever default access control policies it deems appropriate.  After the authorize method has been invoked (passing security tokens for one or more entities e.g. user and/or device), the server is similarly free to implement whatever authorization / access control policies it *deems appropriate* based on the identity of the entities represented by the tokens.  So passing tokens that are associated with identified (authenticated) entities changes the level of authorization for the connection that was used to pass the 'authorize' request.

aShinjiroUrata, 2017-04-07:
Thanks for the comment, Kevin-san!

aShinjiroUrata, 2017-04-14:
Since I already made PR and merged, I close this issue.

closed 2017-04-14 by aShinjiroUrata