w3c/automotive #148: G) 5. Subscription Object > EXAMPLE3: `onExpire` necessary?

aShinjiroUrata, 2017-03-21:
Seprarated from #137

====================
@aShinjiroUrata wrote at 2017 Feb22

G) 5. Subscription Object > EXAMPLE3

There is `onExpire` description in VIAS but not in VISS. 
(I think, this is something related to authentication's TLL.)
Is `onExpire` needed?

====================
@pkinney wrote at 2017 Mar8

G) It might be possible to handle expiration of authorization via a general .onError, but would be up for any proposed more specific handling of this.

====================
@aShinjiroUrata wrote at 2017 Mar21

Not yet reached to conclusion.
Since `onExpire` is not described in VISS, just simply delete?
Or, describe `onExpire` as optional and leave this as implementer's freedom?

peterMelco, 2017-03-22:
I think that deleting onExpire for and handle the error through onError is a better choice.

aShinjiroUrata, 2017-03-22:
Thanks Peter-san,
> I think that deleting onExpire for and handle the error through onError is a better choice.
This is OK to me. Implementer can create onExpire if he want and it is freedom.

If there's no other opinion, let's update the spec as you wrote and close this issue.

aShinjiroUrata, 2017-03-30:
Fixed.

closed 2017-03-30 by aShinjiroUrata