w3c/automotive #156: Whether VIAS is high level API or Websocket based APIs?

wonsuk73, 2017-03-27:
This issues is raised by @aShinjiroUrata. Current Initial draft of VIAS is designed on top of websocket APIs. So doesn't care about other underline protocols like HTTP REST. Is it right way to go? or VIAS might need to be more high level interface regardless of protocols?

aShinjiroUrata, 2017-03-31:
Hi,

This issue was originally posted from Hirabayashi-san at 
https://github.com/w3c/automotive/pull/137#issuecomment-282141366

and I think other members such as Wonsuk-san, Paul-san are also aware.

The background is, in the charter document written about VIAS spec as
https://w3c.github.io/automotive/charter-2016/
> In addition, a higher level API which is intended to use the service will be specified for facility to web developers.

The current VIAS doc is written assuming ...
- VIAS uses WebSocket for communication
- VIAS is implemented as JS library

From one point of view, as a high-level API ...
- VIAS should be agnostic of communication method
- VIAS should not define implementation method
 (could be JS lib, native implementation in Browser, etc.)

I'd like to collect opinions from members.

I think the actual options we can take should be

1)From current VIAS, remove WebSocket, JavaScript library related description
 and rewrite as high-levele API.
- Current VIAS is well organized as WebSocket/ JS library based spec.
- I suppose, probably better to keep WebSocket/JavaScript related descriptions
  at somewhere for people creating JS lib with WebSocket communication.

2)In VIAS document, add 'normative', 'non-normative' words to show
 WebSocket, JavaScript library related description are non-normative.
- Not sure if it could possible to cleanly separate by adding
  'normative', 'non-normative'.

3)Use current VIAS mostly as it is as a version for CR.
- As WebSocket/JavaScript based spec, well organaized.
- Since current version of VISS is assuming WebSocket communication,
  therefore assuming WebSocket communication in this version of VIAS
  might be acceptable.
- Probably we can modify VIAS as more generic and connection agnostic
  in later version.

Please let me know your opinion.

rstreif, 2017-03-31:
In my opinion, to be consistent about that VISS and VIAS are intended to be complementary and working together VIAS needs to define the northbound interface to web applications as JavaScript and the southbound interface as a web socket interface to a VIS server.
I strongly suggest that we are explicit about web socket as the transport and define the JSON protocol. The spec needs to be coherent and implementable and provide interoperability. We should not encumber the spec with additions and modifications that might happen in the future such as the addition of other protocols. When this is happening then we can revise this specification.
In a call that I had with @tguild a while ago, I verbally laid out a path of separating the data domains such as vehicle information (signal data), media, location, etc. from the transports, protocols and encoding. When it is time then the API parts of VISS and VIAS should be consolidated into a single VIS spec which the transport, protocol and encoding pieces of the specification should be consolidated into a web socket protocol spec. The we can add other domains and we can other protocol specifications with the goal that any domain can be used with any protocol spec. But for now we should focus and make work what we have.

Hirabayashi-KDDI, 2017-04-01:
It seems to me that JavaScript library is one of applications, and out of scope in W3C's recommendation.
The current draft of VIAS should be attached as an appendix or a guideline in VISS, for interworking between VISS and VIAS of a high-level API.
FPWD of VIAS as a high-level API will be easily made by just removing parts and descriptions relating to the "connection" from the current draft.
My question is who will make WRT in conformity with VIAS.
We are not still certain whether we will have 2 (two) implementation experience or not.
If implementation testing by using a polyfill for VIAS would be acceptable as Automotive WG and W3C, the implementation issue will be resolved.

aShinjiroUrata, 2017-04-12:
For recording purpose, copy the discussion from 2017/04/04 conference call:

> **Urata**: there is an opinion that VIAS could abstract itself away from web socket
> 
> **Paul**: my understanding was initially we wanted this higher level API to resemble the earlier WebIDL spec
> ... that has shifted
> ... we can move it out later after CR
> ... and defer on decision for now. it can be moved into an appendix
> ... that was Hira's suggestion, to have VIAS as an appendix later
>  
> **Ted**: it is already spelled out as a separate entity in the charter so probably should stay that way
>  
> **Wonsuk**: we can have two variations or really try to figure out how to make a more convenient api for developers
> ... one way to simplify is to poll the WG participants to see which they are planning on implementing
> 
> **Rudi**: we should try to deliver both at the same time
>  
> **Paul**: they can move as separate entities, having the service available sooner than the high level
>  
> **Ted**: let's keep VISS on schedule, these minutes and email from one of the chairs for people
> ... to review the issues they raised or respond to try to wrap them up with final suggestions

aShinjiroUrata, 2017-04-19:
In 2017/04/18 conf call we had discussion about this issue.
https://www.w3.org/2017/04/18-auto-minutes.html

Conclusions are
- remove description like "should be implemented as JavaScript library". 
  Not define how VIAS should be implemented with.
- remove description about "direct correlation with VISS and sockets".

Below is the discussion

> [[Whether VIAS is high level API or Websocket based APIs?]]
> 
> **Urata**: if this as a high level API then it should be more abstract
> ... Rudi's comment was that VIAS is a counterpart to VISS and should be described concretely, using web socket
> 
> **Gunnar**: agree we do not need to mention explicitly the Javascript library and that is consistent with your WebIDL definitions
> 
> **Urata**: if it is a library definition then it would be considered an application
> 
> **Peter**: I do not see how it would be defined as an application
> ... a library is not an application
> ... I agree it should not be defined as a JS library
> 
> **Urata**: yes, it can be implemented as a JS library but the spec does not define how it should be implemented
> ... we should remove the phrases in the spec that call it a JS library
> 
> **Gunnar**: isn't it possible to check what is normally done in W3C?
> 
> **Peter**: usually it would be WebIDL at W3C but we are also not the usual group
> 
> **Rudi**: correct, it would be a definition of an extension or to be implemented directly in a browser. we have not received support from browser vendors and took a different route
> 
> (service/sockets)
> 
> **Rudi**: agree we should remove JS library and I am OK with removing the   direct correlation with VISS and sockets
> ... a VIAS compliant client would most likely need to implement against VISS but it does not need to be explicit
> ... presently we only have 1 server definition, VISS
> 
> **Urata**: in the current VIAS document there are connect and disconnect methods
> ... if the underlying connection is something other than web sockets it might be a stateless service without a persistent connection
> 
> **Rudi**: we can clarify connection-less protocol option
> 
> **Urata**: that is a good solution
> ... I will address 156
>

aShinjiroUrata, 2017-04-21:
With PR #182 followin issues are addressed.

- remove description like "should be implemented as JavaScript library".
Not define how VIAS should be implemented with.
- remove description about "direct correlation with VISS and sockets".
- leave connect / disconnect method and add explanation like
"in case of using connectionless network(e.g. HTTP), connet/ disconnect
would be not necessary."

Still "Security and Privacy" part needs to revise according to these change.

aShinjiroUrata, 2017-04-25:
"Security and Privacy" part has been updated.
Close this issue.

closed 2017-04-25 by aShinjiroUrata