w3c/automotive #158: Clarify Authorise Behaviour

acrofts84, 2017-03-28:
Relating to #140, the current authorise section contains a table which may be a source of confusion as it does not align with the WebIDL/JSON Schema. 

Element | Description
------------ | -------------
Path |	The signal path (defined here) the token authorizes. The path may be a branch name or contain wildcards to authorize entire branches.
Actions	|List of actions that the token authorizes for the path. The list contains at least one of the actions getVSS, get, set, subscribe and unsubscribe.
Valid From	|Timestamp in UTC indicating the date and time from which on the token is valid.
Valid Until	|Tmestamp in UTC indicating the date and time until which the token is valid.

What is the intention of this table? I suggest that this should be removed, as it does not align with the WebIDL which states the following:
```
interface authorizeRequest {
    attribute Action    action;
    attribute DOMString tokens;
    attribute DOMString requestId;
};
```
N.B. The "tokens" type should also be changed from "DOMString" to "object".

rstreif, 2017-03-28:
As the spec treats the token as an opaque element it does not attempt to detail the structure of the token at all. The table only shows what minimum information a token should contain. I am not quite sure what the `attribute Action action;` stands for in the `interface authorizeRequest`. Is that the action the client wants be authorized for? That I think must be part of the token as Path, Actions, Valid From, and Valid To need to be kept together.
We may also want to keep the spelling consistent: the code examples and JSON use the AE spelling _authorize_, while the text uses the BE spelling _authorise_.

acrofts84, 2017-03-29:
The authorizeRequest is defined as an object with the following structure, whereby there can be multiple token key-value pairs within the token object:
```
'{"action": "authorize", "tokens": {"authorization": "user_token_value", "www-vehicle-device": "device_token_value"}, "requestId": "103"}'
```
The action is "authorize" to inform the server explicitly that this is an authorization request. The current token structure is _just_ a token without meta-data. As such the information about the path, actions and valid to/from for a given token is not relevant to the process of authorizing the WebSocket channel. It will be able to apply its current tokens and if the security authority has granted the required access with the supplied token(s), the application will be able to access the signals and actions it desires. The server will have to verify this with the security authority.

The tokens are opaque insofar as you do not know their exact format, however we expect that they are just a token, i.e. JWT or OAuth.

In short, the details from the table are pertinent to the generation of the token(s), not the application of the token(s).

I'm happy to discuss this further so we can clarify the desired behaviour. 

I agree with the spelling issue, this should be consistent.

rstreif, 2017-03-29:
Ok, thank you. It makes sense this way. We can remove the table. The only intend of the table was to illustrate what a token should reasonably contain but since this is a spec and the table is not normative it does not add any value.

acrofts84, 2017-04-28:
I have raised #192 for this and already fixed the spelling issue

drkevg, 2017-05-03:
Agree with Adam and Rudi that table should be removed.

acrofts84, 2017-05-08:
Merged changes

closed 2017-05-08 by acrofts84