w3c/automotive #163: W3CClient class should be named more specifically

rstreif, 2017-03-30:
The term W3CClient class introduced in section 4 of the [VIAS](https://w3c.github.io/automotive/vehicle_information_api/vehicle_information_api_specification.htmll) is rather generic. We typically refer to the server as VIS server. Maybe the term VISClient class would work in this context.
We also need to keep in mind that the specification eventually will evolve and that there may be other clients and servers.

aShinjiroUrata, 2017-03-31:
Rudi-san, thanks for the comment.

> We typically refer to the server as VIS server. 
> Maybe the term VISClient class would work in this context.

> I think the heading of section 4 should use the term "Class" instead of "Object".

Both make sense to me.
I will update these.

aShinjiroUrata, 2017-03-31:
The update done.
Close this.

closed 2017-03-31 by aShinjiroUrata