w3c/automotive #171: Handling optional property in JSON schema

wonsuk73, 2017-04-04:
Below is the JSON schema for vssRequest in the current draft and vssRequest consists of 'action', 'path', 'requestId' properties. 'path' is only optional property. Below schema defines all of three properties including 'path' as a mandatory to be described. Then 'path' could have a 'null' value.

I think it is not the best way for our intention.

{
    "$schema": "http://json-schema.org/draft-04/schema#",
    "title": "VSS Request",
    "description": "Request metadata describing the potentially available VSS tree",
    "type": "object",
    "required": ["action", "path", "requestId"],
    "properties": {
        "action": {
            "enum": [ "getVSS" ],
            "description": "The identifier for the getVSS request",
        },
        "path": {
            "$ref": "#/definitions/path",
            "type": ["string", "null"]
        },
        "requestId": {
            "$ref": "#/definitions/requestId"
        }
    }
}

My proposal is like this. VSS Request object consists of three properties but mandatory are 'action' and 'requestId'. So in case of 'path' that could be used when the case is required to use 'path'.

{
    "$schema": "http://json-schema.org/draft-04/schema#",
    "title": "VSS Request",
    "description": "Request metadata describing the potentially available VSS tree",
    "type": "object",
    "required": ["action", "requestId"],
    "properties": {
        "action": {
            "enum": [ "getVSS" ],
            "description": "The identifier for the getVSS request"
        },
        "path": {
            "$ref": "#/definitions/path"
        },
        "requestId": {
            "$ref": "#/definitions/requestId"
        }
    }
}

@acrofts84, What do you think? I will prepare for PRs for applying this to the whole of draft.

rstreif, 2017-04-04:
@wonsuk73, Does this mean if the path is omitted then the entire VSS tree is returned?
I a sense an empty path seems to be implied by `"type": ["string", "null"]`. Consequently, if path can be empty or null then it makes only sense to make it optional.
Makes sense to me.

wonsuk73, 2017-04-07:
@rstreif,
Yes. As you think omitting 'path' property means to ask  the whole of VSS file. When the client want to ask only the some part of VSS tree, client should use 'path' property to indicate the part he want. I think for better understanding it would be good to add examples to getVSS part for both of cases. If we have a consensus on this, I will make a PR for applying it.

rstreif, 2017-04-07:
@wonsuk73 
Thank you. It's all good with me. Please make the changes and create the PR to update the spec.

acrofts84, 2017-04-11:
Thanks @wonsuk73, this looks good to me

acrofts84, 2017-05-10:
Discussed at F2F. See #201

closed 2017-05-10 by acrofts84