w3c/automotive #174: Gap

wonsuk73, 2017-04-05:
Basically this is related with #171 of issues. Regarding to vssRequest WebIDL definition below the 'path' property of vssRequest interface is optional so vssRequest JSON message that doesn't have 'path' property could be allowed to be used in some cases.
~~~~
interface vssRequest {
    attribute Action     action;
    attribute DOMString? path;
    attribute DOMString  requestId;
};
~~~~

But according to the JSON schema definition vssRequest must have 'path' property with 'null' value even the the case when 'path' doesn't have a meaning. So it could make confusion for the implementer.
~~~~
{
    "$schema": "http://json-schema.org/draft-04/schema#",
    "title": "VSS Request",
    "description": "Request metadata describing the potentially available VSS tree",
    "type": "object",
    "required": ["action", "path", "requestId"],
    "properties": {
        "action": {
            "enum": [ "getVSS" ],
            "description": "The identifier for the getVSS request",
        },
        "path": {
            "$ref": "#/definitions/path",
            "type": ["string", "null"]
        },
        "requestId": {
            "$ref": "#/definitions/requestId"
        }
    }
}
~~~~

So it is better to use below JSON schema for vssRequest interface with additional clear sentence for getVSS part of the spec. It should explain the case which doesn't use 'path' property, in order to request the entire VSS files. In addition it need to say the case which does use 'path' property for requesting only some part of VSS file to the server. For these two cases examples also would be helpful for the readers.

~~~~
{
    "$schema": "http://json-schema.org/draft-04/schema#",
    "title": "VSS Request",
    "description": "Request metadata describing the potentially available VSS tree",
    "type": "object",
    "required": ["action", "requestId"],
    "properties": {
        "action": {
            "enum": [ "getVSS" ],
            "description": "The identifier for the getVSS request"
        },
        "path": {
            "$ref": "#/definitions/path"
        },
        "requestId": {
            "$ref": "#/definitions/requestId"
        }
    }
}
~~~~

drkevg, 2017-05-03:
What do people think of making path mandatory and non nullable, and if there is not a path value then specifying empty string ""?

So schema would be:
<pre>
{
    "$schema": "http://json-schema.org/draft-04/schema#",
    "title": "VSS Request",
    "description": "Request metadata describing the potentially available VSS tree",
    "type": "object",
    "required": ["action", "path", "requestId"],
    "properties": {
        "action": {
            "enum": [ "getVSS" ],
            "description": "The identifier for the getVSS request",
        },
        "path": {
            "$ref": "#/definitions/path",
            "type": ["string"]
        },
        "requestId": {
            "$ref": "#/definitions/requestId"
        }
    }
}
</pre>

Believe this is simpler in implementation and the intention is clear from the data that is passed. Look forward to hearing other views and perhaps discussing at the F2F in May

drkevg, 2017-05-03:
Sorry meant to flag on previous post that this topic is also discussed on #171 and #176

acrofts84, 2017-05-10:
Discussed at F2F. See #201

closed 2017-05-10 by acrofts84