w3c/automotive #183: VISClient - No method to register ondisconnect callback

rstreif, 2017-04-21:
VISClient does not provide a method to register an ondisconnect callback. I can call disconnect(callback) to close the connection from the client side but there is no way to react if the connection is closed by the server or interrupted by other means. There should be a VISClient method allowing to register a callback for disconnection. That could be separate or a parameter to the connect method e.g. VISClient.connect(connectCallback, disconnectCallback).

aShinjiroUrata, 2017-04-24:
Hi Rudi-san,

> VISClient does not provide a method to register an ondisconnect callback. 
> I can call disconnect(callback) to close the connection from the client side
>  but there is no way to react if the connection is closed by the server or interrupted by other means. 

Regarding this part, I have not changed from Powell-san's original and I understand that, 
 it is possible to react to disconnection from server-side by onDisconnect handler.
```
var cli = new VISClient( /**/);
cli.onDisconnect = (_cli) => { /* disconnect handler */ };
cli.connect(  (_err) => { /**/ } );
```

This, having `onConnect`, `onDisconnect` handler style is similar to WebSocket interface
which has `onopen`, `onclose` and I think basically OK.

But another thing I noticed is 
```
    attribute Function? onConnect;
    attribute Function? onDisconnect;
```
handlers and callback functions of 
```
    void? connect(connectCallback callback);
    void? disconnect(disconnectCallback callback);
```
are basically similar things and probably we should remove one of these.

I'll consider about this.
If you have comment, please let me know.

aShinjiroUrata, 2017-04-25:
I propose below idea.

- remove, `onConnect`, `onDisconnect`, `onError`, `onMessage`
  - elliminate onXXXX handlers for style coherency.
- modify connect() method as below. 
  - adding errorCallback instead of  deleting `onError`
  - Regarding `onMessage`, I simply deleted because it is not an essential method.
```
connect(connectCallback, disconnectCallback, errorCallback);  
```
- modify disconnect() method to
```
disconnect();  //disconnect callback is already registered in `connect()` method
```

aShinjiroUrata, 2017-04-26:
Hi @rstreif san

I created PR #187 as I wrote in previous message.
Coud you review this?

rstreif, 2017-04-26:
@aShinjiroUrata Thank you, as commented on PR #187 that looks good to me.

aShinjiroUrata, 2017-04-27:
Thanks for the review !
I have merged the change.
Please close this if you are OK.

closed 2017-05-11 by aShinjiroUrata