w3c/automotive #198: VIAS Comments

acrofts84, 2017-05-09:
@aShinjiroUrata, thank you for your work on the VIAS spec. So far it looks really good. 

I have a few comments. I'm happy to raise them in individual issues if they need more in depth discussions.

**1 Authorise method**
<pre>
void  authorize(DOMString tokens, AuthorizeCallback callback);
</pre>
I think 'tokens' may need to remain a JSON object to allow full flexibility. How do we define multiple tokens in this format? Can we specify the name of the token User-Token: "asd-asd-asd" and Vehicle-Token: "sdf-sdf-sdf"? 
            
**2 Naming Subscribe Callback**
Should we call this a SubscriptionCallback, since subscribeCallback implies on subscribe, rather than on subscription notification?
<pre>
        void  subscribe(DOMString path,
                    SubscribeCallback callback,
                    optional SubscribeFilters filters);
</pre>

**3 WSS Protocol**
In Example 2 you have specified the protocol as ws rather than wss, but non encrypted communication is not supported.

**4 Multiple  Callback**
The Authorise callback will receive either a TTL or an error, never both. However your definition states that it will always receive TTL and VISError. Likewise the GetVSSCallback will only receive either a VSS object of a VISError, and similar for the other callbacks. It will be easier to decouple the success path and the error path by having separate callbacks.
<pre>
        callback AuthorizeCallback = void (unsigned long TTL, VISError error);
</pre>
Do we need to change the method definition to this, or similar for the other callbacks:
<pre>
        void? connect(ConnectCallback connectCallback,
                      DisconnectCallback disconnectCallback,
                      ErrorCallback errorCallback);
        void  authorize(DOMString tokens, AuthorizeCallback callback, ErrorCallback errorCallback);
        void  getVSS(GetVSSCallback callback);
        void  get(DOMString path, GetCallback callback, ErrorCallback errorCallback);
        void  set(DOMString path, any value, SetCallback callback, ErrorCallback errorCallback);
        void  subscribe(DOMString path,
                        SubscribeCallback callback,
                        ErrorCallback errorCallback,
                        optional SubscribeFilters filters);
        void  unsubscribe(VISSubscription subscription,
                          UnsubscribeCallback callback,
                          ErrorCallback errorCallback);
        void? disconnect(DisconnectCallback connectCallback, ErrorCallback errorCallback);
</pre>

**5 VISClient - Nullable Methods**
I appreciate the thought behind connect and disconnect being nullable methods, however I'm not sure this is the best way to approach the issue, since this could cause a runtime error. I think a better approach is for an ErrorCallback to be called with something like a "501 Not Implemented"
<pre>
    void connect(ConnectCallback connectCallback,
                  DisconnectCallback disconnectCallback,
                  ErrorCallback errorCallback);
    void disconnect(ErrorCallback errorCallback));
</pre>

aShinjiroUrata, 2017-05-10:
Hi @acrofts84 san
Thank you for the comments!

> 1 Authorise method
> How do we define multiple tokens in this format?

My Idea was, any JSON object can be stringified to a string and so DOMString can work. 
But as it is misleading, better to change to `any` or `object`.

> 2 Naming Subscribe Callback
> Should we call this a SubscriptionCallback, since subscribeCallback implies on subscribe,
> rather than on subscription notification?

Actually, this callback has both functions of on-subscribe and on-subscriptionNotification.
Which means, not only the value of subscribed item but `subscriptionObject` is also returned
 by this callback.
Some time before, `subscribe()` had two calbacks like below. I thought two callbacks are redundant
 and changed to one callback style.
```
    void subscribe(DOMString path,
                   subscribeCallback callback,
                   subscriptionCallback callback,
                   optional SubscribeFilters filters);
```
I got another input from KDDI which is, want to receive subscriptionObject or subscriptionID 
 **synchronously** like below since from web developer point of view, this is more familiar.
(For developer's convenience, I understand this would be better)
```
    VISSubscription subscribe(DOMString path,
                   subscriptionCallback callback,
                   optional SubscribeFilters filters);
```
I'd like to discuss on this in F2F.

> 3 WSS Protocol

Thanks. I'll fix this.

> 4 Multiple Callback

I understand the point and I'm OK to change like this.

> 5 VISClient - Nullable Methods
> I think a better approach is for an ErrorCallback to be called with something like a "501 Not Implemented"

I agree. This looks good approach.

aShinjiroUrata, 2017-05-13:
Birmingham F2F results memo.

> 1 Authorise method
Simply fix as describe in previous post.

> 2 Naming Subscribe Callback
> 4 Multiple Callback
- Issue about callback is not easy to decide solution.
  Leave this issue as open and publish FPWD first
  and continue discussion.
- There was a proposal to use Promises instead of callback.
  However, we do not change to promises style right away since,
  - continuous data push of `subscribe` method could not be realized
     by Promises style.
  - If some methods are not achivable with Promises, convert some method to Promises
     and leave some method as callback style is not good.
     In that case, leave all methods as callback style would be better.

> 3 WSS Protocol
- From PatricB opinion, mandating `wss` will cause problem such as
   when certificate in embedded device expired, some device may not 
   be able to update the certificate.
- This is not a VIAS issue but a VISS issue.
- Regarding VIAS, just modify to `wss` to follow description in VISS and 
   continue this disscusion as VISS issue. 
- Also, this is not only about W3C automotive issue but more common topic.
  There is W3C https local Community Group which is tackling related issue. 
  We should watch progress in this CG.

> 5 VISClient - Nullable Methods
Fix as describe in previous post.

aShinjiroUrata, 2017-05-16:
By PR #202 , fixed above issue 1, 3, 5.

2, 4 are remain unchanged.

aShinjiroUrata, 2017-05-17:
By PR #205, followed "4 Multiple Callback" proposal to have success and error callbacks
 since the advise looks reasonable.

aShinjiroUrata, 2017-09-05:
This discussion has already finished.
I close.

closed 2017-09-05 by aShinjiroUrata