w3c/automotive #210: Change "getVSS" to "discover"

acrofts84, 2017-05-25:
In order to further abstract the data model from the access methods, we should rename the "getVSS" method/action to the "discover" method/action.
- This is familiar parlance within the IoT and will be more familiar to an implementer.
- This ensures that the spec can still be logically applied to new data models, including but not limited to the VSS.

aShinjiroUrata, 2017-05-26:
Hi @acrofts84 san

I think abstraction of 'getVSS' is good idea.

By the way, although I am not very familiar with IoT specs, this word "discover" is appropriate for this use case?
In w3c wot, "Discovery AP" is to discover IoT "things" which is like discovering devices.
http://w3c.github.io/wot/architecture/wot-architecture.html

If the word "discover" is natural in this use case, I'm OK to change.

PeterWinzell, 2017-05-30:
I agree "discover" is better

drkevg, 2017-05-30:
Although we like 'discover', Adam and I think Urata-san raises a good point and that using 'discover' could be confusing if we use it to discover metadata and e.g WoT use it to discover services. Would it be ok if we replaced getVSS with getMetadata instead (we also considered getSchema, but the method call returns an object with metadata rather than a schema definition, so getMetadata seems clearest)

peterMelco, 2017-05-30:
Ok, I then change my mind :) getMetaData is more clear, so I am fine with that.

acrofts84, 2017-05-31:
I also agree with getMetadata

aShinjiroUrata, 2017-06-02:
Excuse me for late reply.
I'm OK with the word 'getMetadata'.

I looked more at OIC Core spec and looks, word 'discover' is used not only for 'Endpoint discovery',
  used for 'Resouce discovery' as well.
https://openconnectivity.org/developer/specifications

So, I thought something like 'discoverResouce' could be another candidate
 but it is not something like commonly used word.

So, in our case, I suppose 'getMetadata' matches well.

rstreif, 2017-06-02:
I agree that just 'discover' is too generic and somewhat overloaded. @acrofts84, looking at OIC is a good idea because of the interaction with them and making things consistent even across different applications makes a lot of sense. 'discoverResource' has my vote but I am ok with 'getMetadata' although it's a little too generic for my taste. :)

peterMelco, 2017-09-05:
So I will close this one since this is now changed in the spec to getMetaData

closed 2017-09-05 by peterMelco