w3c/automotive #214: Duplication of host,protocol,port in VISClient and VISClientOptions

anawhj, 2017-06-22:
I found something strange while I read the VIAS specification carefully.

In VISClient Interface, 'host', 'protocol', 'port' attributes are defined twice in VISClient and VIS ClientOptions. If the attributes should be used in only creating an instance, the definitions in VISClient should be eliminated. If not, the definition of attributes in VISClient should be specified in the VISClient scope independent of VISClientOptions.

If I miss something about my understanding, please let me know the correction. I guess @aShinjiroUrata could handle this issue :)

aShinjiroUrata, 2017-06-22:
Hi @anawhj san,
Thanks for the comment!

> In VISClient Interface, 'host', 'protocol', 'port' attributes are defined twice in VISClient and VIS ClientOptions. 

The 'host', 'protocol', 'port' in `VISClientOptions` are for setting values at instance creation.
Those in `VISClient` are just for read the vaues (so, defined as `readonly`).

The reason 'host', 'protocol', 'port' are added in VISClient is, it would be inconvenient if those valuse
 are not available to read after the instance is created. 

What do you think?

anawhj, 2017-06-22:
Hi Urata san,

I'm Hyojin Song working in LG Electronics, and I'm simply looking into a couple of specs in Automotive WG. I could well understand your explanation in which the attributes are used for getting the value as `readonly`.

In that case, I would suggest that the attributes are defined in the scope of `VISClient`, and no need to define `VISClientOptions `interface. In addition, the constructor's parameters for `VISClient` would be mandatory, not optional depending on the usage.

You can see an example to review this issue in the following link.
https://www.w3.org/TR/dom/#interface-event

aShinjiroUrata, 2017-06-22:
Hyojin-san,

Thanks, I understand your point.

(1)style
With the current spec's style(style A), the code is like
```
var client = new VISClient( { 'host': 'wwwivi', 'protocol': 'ws', 'port':80 } );
```
and with your suggesting style (style B),
```
var client = new VISClient( 'wwwini', 'ws', 80);
```
I think both styles are found in other W3C specs.
style A example:
https://www.w3.org/TR/webrtc/#interface-definition (WebRTC  RTCPeerConnection)
style B example (as you wrote):
https://www.w3.org/TR/dom/#interface-event (DOM4 Event interface)

When there are many arguments, I suppose style A is tend to be selected.
In our case, I think, both style would work, and which means
,personally, I don't find a strong need to change to style B.
What do you think?

As other reason, we knew that the first VIAS implementation will be created
 as JavaScript library and in JavaScript case, style A is easy, flexible and convenient.

(note: VIAS itself is not a spec dedicated for JavaScript implementation and someone would be able to
 implement it in browser with WebIDL and native code.)

(2)optional
The reason why `VISClientOptions` argument is optional, (in other word, 'host', 'port', 'protocol'
 are optional) is, this VIAS spec just defines API interface and does not assume what network
 connection will be used in the lower layer.

In case of using WebSocket, 'host', 'port', 'protocol' will be necessary but if other connection
 (e.g. http, mqtt or native IPC) is used, these parameters might not be needed.
Therefore, these parameters are described as optional and I think 'mandatory' does not fit in this case.

By the way, since the first candidate connection type is WebSocket, 'host', 'port', 'protocol' are
 defined in VISClietnOptions interface. 
In the future version of VIAS, VISClientOptions' member attributes could be added to support
 other network type.

Does this explanation make sense to you?

anawhj, 2017-06-23:
Thanks a lot for kind explanation and sorry for taking your time to explain in detail.
I can exactly understand it and mostly agree with your opinions.

Honestly, I can find this concerns while fixing warnings and errors reported by Respec. (#215)
I would suggest my opinions as some candidates to solve this issue as follows.

**1. VISClientOptions style**
WebRTC you mentioned could be a good example to review this case. I also prefer style A, but the definition of `VISClientOptions` IDL would be specified as a dictionary rather than a interface same as [`RTCConfiguration`](https://www.w3.org/TR/webrtc/#dom-rtcconfiguration) case. The change from interface to dictionary can resolve the Respec warning regarding specifying each attribute definition as mandatory. I can find another references like [`UDPSocket`](https://www.w3.org/TR/raw-sockets/#interface-udpsocket), [`TCPSocket`](https://www.w3.org/TR/raw-sockets/#interface-tcpsocket) in the [Socket API](https://www.w3.org/TR/raw-sockets/) Spec though it was deprecated as a note.

In addition, get/set methods(e.g. getVISConnection) could be used to access the attributes(host, protocol, port) instead of providing three `readonly` attributes. Using set() method, re-connection could be operated to connect to the VIS.

BTW I'm not sure style A is more convenient, flexible, and easier than style B as you wrote. Even if many arguments are defined as parameters of a function, it can be easily handled using [`arguments`](https://www.w3schools.com/js/js_function_parameters.asp) object within the function's local scope. So the level of difficulty between style A and style B would hardly make difference as my personal viewpoint.

**2. optional**
As I understand your explanation, the reason of the use of 'optional' would be for considering other lower layers other than WebSocket so that we would consider an extensibility of `VISClientOptions` to support various underlying protocol such as HTTP(Restful), MQTT. However I think 'optional' would have no correlation with the extensibility.

For example, If we specify 'optional' to `VISClientOptions`, the constructor of `VISClient` should be executed properly without `VISClientOptions` parameter, but it seems a little strange in my opinion. If `connect` method supports a way to connect to VIS with 'host', 'port', 'protocol', the constructor of `VISClient` without parameters would be reasonable.

aShinjiroUrata, 2017-06-27:
Hyojin-san,

Excuse me for late response and thanks for explanation.
I understand your comment's background better.

> 1. VISClientOptions style
> I also prefer style A, but the definition of VISClientOptions IDL would be specified as a dictionary rather than a interface same as RTCConfiguration case. 

I agree that `dictionary` is correct.
 
> In addition, get/set methods(e.g. getVISConnection) could be used to access the attributes(host, protocol, port) instead of providing three readonly attributes. Using set() method, re-connection could be operated to connect to the VIS.

I think I understand your idea. 
What will be the advantage of the strategy you wrote?

> BTW I'm not sure style A is more convenient, flexible, and easier than style B as you wrote. Even if many arguments are defined as parameters of a function, it can be easily handled using arguments [...]

I think this includes my personal preference.
As you wrote, it is possible to handle many arguments with `arguments` object.
In that case, 
- We have to be careful about the order of arguments. 
  Also, we have to handle with name like arguments[i] and need to memorize
  such as ' arguments[1] is protocol and arguments[2] is port'.
- In addition, even if no need to specify argument 1, we have to write like
```
var client = new VISClient('127.0.0.1, '', '8080')
```

In case of style A, 
- we can use arguments by name like `arg.protocol`, `arg.port`
- Also, we just skip unnecessary arguments like below and no need to care about
  order.
```
var client = new VISClient({ host:'127.0.0.1', port: '8080'}) 
```
I personally feel style A is easy and convenient.
However, in case of only three arguments, I think there is not much difference.

> 2. optional
> As I understand your explanation, the reason of the use of 'optional' would be for considering other lower layers other than WebSocket so that we would consider an extensibility of [...]

Actually, it was bit difficult to find a good way of description with having WebSocket, HTTP, and
 other method under consideration as underling communication path.
In case of using WebSocket communication as underlying communication, host, protocol,
 port will be necessary.
However, for example, if VIAS is imeplemented into a browser as native code and the communication path is something like IPC, arguments like host, protocol, port will not be necessary.

The use of 'optional' in this place is, 
- if a VIAS implementation is using WebSocket communication, 'host', 'protocol', 'port' are necessary
- if a VIAS imple using IPC-like communication, 'host', 'protcol', 'port' are not necessary
as such, necessity of arguments are different by implementation. 
In a case, arguments are necessary and another case, no arguments are required
 and I used the word 'optional' to describe this situation. 
But as you say, this usage of might not be standard usage of 'optional'.

If you have better idea to describe this situation in WebIDL, I welcome any suggestions.

> If connect method supports a way to connect to VIS with 'host', 'port', 'protocol', the constructor of VISClient without parameters would be reasonable.

Your idea is something like below?
```
[Constructor()]
interface VISClient {
VISClientOptions  getVISConnection();
void  setVISConnection(host, protocol, port);
void  connect(host, protocol, port);
void authorize(...);
void getVSS(...);
...
...
}
```

anawhj, 2017-06-28:
I could sum up the issues in this thread as the following list.

1. The type of `VISClientOptions` changes from a interface to a dictionary. `VISValue` and `VISError` would be similar cases with that, so each type of the objects could change from a interface to a dictionary as well.

2. Regarding the advantage of get/set methods for host/protocol/port, the methods could make the structure of `VISClient` interface simplify through hiding the attributes and providing the unified methods. At a first glance, it seems duplicated definitions about `VISClientOptions` object and host/protocol/port readonly attributes. I would suggest that we define `VISConnections` dictionary(the terminology could be changed later) so that we can use `VISConnections` as the parameter of constructor in `VISClient` and as a readonly attribute to get the values about host/protocol/port. This structure could be described like this.

```
// VISClient Interface IDL
[Constructor (VISConnection conn)]
interface VISClient {
  readonly attribute VISConnection conn;
  void connect(ConnectCallback connectCallback,
               ErrorCallback errorCallback);
...
dictionary VISConnections {
  attribute DOMString?      host = "wwwivi";
  attribute DOMString?      protocol = "wss";
  attribute unsigned short? port = "443";
};
```

3. With your explanation, I feel like the object-based style would be better than the arguments style.

4. Regarding `optional` keyword, I understood your comment about the IPC communication and it seems a reasonable situation. In that case, we could consider function overloading so that `'var instance = new VISClient(VISConnections conn)'` makes a connection of underlying protocol such as WebSocket and HTTP, and `'var instance = new VISClient()'` makes a connection of IPC communication which doesn't need a specific target address. At this stage, we could only focus on the former one to connect to VIS through WebSocket. At some time, we'll most likely be able to easily extend the function to support the latter one. Surely, we can define both ones without `optional` keyword in the spec now.

5. On second thought, we can refresh the web application in the web browser or web runtime to reconnect the VIS, so the connect method might not be necessary for now.

aShinjiroUrata, 2017-06-29:
Hi Hyojin san,
Thanks for further comments!

> 1. The type of `VISClientOptions` changes from a interface to a dictionary. `VISValue` and `VISError` would be similar cases with that, so each type of the objects could change from a interface to a dictionary as well.

Yes, I agree. We should change to `dictionary`.

> 2. Regarding the advantage of get/set methods for host/protocol/port,
 the methods could make the structure of `VISClient` interface simplify
 through hiding the attributes and providing the unified methods. 

Thanks, I understand the advantage.
By the way, I think get/set is popular method in OO languages such as Java,
 but in W3C specs, this get/set method does not look like a common practice.
Don't you think so?

In addition, in WebIDL definition, member attributes of an interface(=class) 
 are not necessarily `private`, so I suppose, the situation is different from
 class of OO language.

> At a first glance, it seems duplicated definitions about `VISClientOptions`
 object and host/protocol/port readonly attributes. 

I understand your point but I suppose
 we don't need to call this 'a duplicate definition'
 since `VISClientOptions` is defined only for usage as VISClient constructor argument.

In most case, VISClientOptions is used as just a JSON literal like below.
```
var client = new VISClient({'host':'wwwivi', 'protocol':'wss', 'port':'443'});
```
If the same values are stored in two different places, I think we sould 
 make it as one,  but in this case VISClientOption object is just a temporal value
 and I think we can leave it.

Also, WhatWG Fetch API's 'Request interface' and 'RequestInit' dictionary are doing
 'duplicate-like' definition. I see this kind of description in some more places.
https://fetch.spec.whatwg.org/#request-class

> I would suggest that we define `VISConnections` dictionary(the terminology could be changed later)
 so that we can use `VISConnections` as the parameter of constructor in `VISClient`
 and as a readonly attribute to get the values about host/protocol/port.
> This structure could be described like this.
>
> ```
> // VISClient Interface IDL
> [Constructor (VISConnection conn)]
> interface VISClient {
>   readonly attribute VISConnection conn;
>   void [...]

Thanks for suggestion.
I see this as, 

- `VISConnections` is just a change of name from `VISClientOptions` and added default values, 

This is OK to me.

- removed `optional` from VISClient constructor argument.

This is discussed in below part.

- replace three attributes(=host, protocol, member) with one attribute(=VISConnectiono conn)
  and VISCllient's interface is simplified.

Although, reducing attributes from 3 to 1 is good simplification, on the other hand,
 to look at other specs, many of them are not very much worry about number of attributes,
 and so, in our case, I personally think, three attributes would be acceptable.
(e.g.  WebRTC: RTCPeerConnection interface
https://www.w3.org/TR/webrtc/#interface-definition )

> 3. With your explanation, I feel like the object-based style would be better than the arguments style.

Thanks. Then let's keep the 'style A' style.

> 4. Regarding `optional` keyword, I understood your comment about the IPC communication
 and it seems a reasonable situation. 
> In that case, we could consider function overloading so that
   `'var instance = new VISClient(VISConnections conn)'`
 makes a connection of underlying protocol such as WebSocket and HTTP,
 and
   `'var instance = new VISClient()'`
 makes a connection of IPC communication which doesn't need a specific target address. 
> At this stage, we could only focus on the former one to connect to VIS through WebSocket. 
> At some time, we'll most likely be able to easily extend the function to support the latter one. 
> Surely, we can define both ones without `optional` keyword in the spec now.

As for the 'overloading' idea, I look up at WebIDL definition,
 it looks ok to have multiple constructor definition.
https://heycam.github.io/webidl/#Constructor  Example50.

So, it would be better to write like below than using `optional`.
The first constuctor is for no-argument case.
What do you think?

```
[Constructor,
Constructor (VISClientOptions options)]
interface VISClient {
...
```

Thanks for the suggestion!

> 5. On second thought, we can refresh the web application in the web browser or web runtime
 to reconnect the VIS, so the connect method might not be necessary for now.

I understand this is about eliminating `connect()`, `disconnect()` methods.

Personally, I think this is not good because this restricts API user's freedom.
I think, connect/disconnect should be possible at the timing API user's want.
Especially, recent SinglePageWebApplication does not refresh browsers for long time.

Also, we had discussion in Burmingham F2F that
- in case of HTTP(S) as underlying communication, connect/disconnect is 
not necessary, different from WebSocket case.
And our conclusion at that time was, as written in spec
 "In such case, this method could be implemented as empty function...".

Thank and regards,

aShinjiroUrata, 2017-07-04:
Hi Hyojin san
I have already fixed the `dictionary` issue by PR #218 (already merged).
For other points, I'm creating another PR.

anawhj, 2017-07-05:
Good to see the PR in which our discussion was properly applied in VIAS specification.

`RTCPeerConnection` interface as you provide could be a good example to review the usability of get/set methods in `VISClient` interface. If `RTCPeerConnection` instance is created without a parameter of the constructor like "_new RTCPeerConnection()_", the object is almost empty and it could connect to sequence of iceServers using `setConfiguration()` with RTCConfiguration parameter that would be the parameter of the constructor as well. Otherwise if `RTCPeerConnection` instance is created with a parameter of the constructor like "_new RTCPeerConnection(RTCConfiguration conf)_", the object would have the IceServers connections at the time of object creation. Besides, the parameter of the constructor has a specific type like RTCConfiguration where several attribute were defined to properly connect to IceServers.

So I have supposed that the specific type named VISConnection so that it could be used as a parameter of VISClient constructor as well as get/set methods or connection readonly attribute to read host/protocol/port data. I agree your opinion on get/set methods (though we can see many cases in W3C specifications(e.g. [RTCPeerConnection](https://www.w3.org/TR/webrtc/#interface-definition) interface), so we could consider a `connection` readonly attribute with a VISConnection type to check host/protocol/port information. Surely `Request` interface in Fetch API is good example, so I agree the separated readonly attributes approach. I'm wondering how other people think about this.

I think your proposal for defining multiple constructors seems no problem. In addition we can add a note to explain the usage of `VISClient` without any parameter of the constructor for IPC communication.

Regarding `connect()`, `disconnect()` methods I totally agree your comments. I intended to define the methods with host/protocol/port as parameters.

aShinjiroUrata, 2017-07-07:
Hi Hyojin san,

I have created PR #220 which is mostly for cleaning up purpose.
- fix remaining ReSpec errors
-- to do this, relocate some sentences
-- added subsctions
-- added tables
-- added dfn, data-dfn-for, data-link-for
- change VISClient's constructor to overloading style

This PR #220 doesn't include the following issues you pointed
- use `VISConnection` for VISClient constructor paramter
- add get/set method of VISConnection

I'd like to consider these points after this clean up.
I hope you can review PR #220.

Thanks and regards,

aShinjiroUrata, 2018-01-08:
Since the related PR is already merged, close this.

closed 2018-01-08 by aShinjiroUrata